nanoframework/Home #1109: M5Stack LED doesn't initialize GPIO leading to errors

jacobguilbeau, 2022-08-02:
### Library/API/IoT binding

M5StickCPlus

### Visual Studio version

VS2019 (16.11.16)

### .NET nanoFramework extension version

2019.11.0.23

### Target name(s)

M5StickCPlus.led.Toggle()

### Firmware version

1.8.0.458

### Description

Trying to run a simple one line statement to toggle the LED fails with CLR_E_NULL_REFERENCE

Code trying to run:
`M5StickCPlus.Led.Toggle(); `

Produces this output:
`The thread '<No Name>' (0x2) has exited with code 0 (0x0).
    ++++ Exception System.NullReferenceException - CLR_E_NULL_REFERENCE (1) ++++
    ++++ Message: 
    ++++ nanoFramework.M5Stack.M5StickCPlus::get_Led [IP: 000e] ++++
    ++++ NFApp2.Program::Main [IP: 0004] ++++
Exception thrown: 'System.NullReferenceException' in nanoFramework.M5StickCPlus.dll
An unhandled exception of type 'System.NullReferenceException' occurred in nanoFramework.M5StickCPlus.dll `

This is due to the led object depending on a gpio internal object that has not been created.  Getting the GPIOController forces the internal object to be created leading to the led object now working correctly.

### How to reproduce

1. Start new nanoframework project
2. Try and do anything with M5StickCPlus.Led

### Expected behaviour

I expect the led to toggle without exception

### Aditional information

Problem is here:
https://github.com/nanoframework/nanoFramework.M5Stack/blob/31c8cf98ba5b458485d41235fe7f0ce50681b955/nanoFramework.M5StickCommon/M5StickCBase.cs#L28

led get just returns _gpio. and doesn't check for null or create a new gpio.

josesimoes, 2022-08-02:
This is using a lazy load pattern. It's been fixed in several other M5stack devices.
Can you take a look and submit a PR with a fix? 😁 That would be great!

jacobguilbeau, 2022-08-02:
Yes!  I'll look at what the other devices are doing here and duplicate.

jacobguilbeau, 2022-08-02:
Looks like I can replace the usage of _gpio in the leds and buttons with GPIOController and this will instantiate it if needed - does that sound ok?

josesimoes, 2022-08-02:
Better keep the existing organization, there are shared classes (for example) and other code using the gpio controller and pins.
Check nanoframework/nanoFramework.M5Stack#193 for insipiration.

jacobguilbeau, 2022-08-02:
Yes, I see what you are doing there.  I'm saying that replacing:
` public static GpioButton Button
        {
            get
            {
                if (_button == null)
                {
                    _button = new(39, _gpio, false);
                }

                return _button;
            }
        }`

With
` public static GpioButton Button
        {
            get
            {
                if (_button == null)
                {
                    _button = new(39, GpioController, false);
                }

                return _button;
            }
        }`

seems to be the proper way to go, no?  That way it uses the loader properly.  There are some files, like M5Stack.Fire.cs where _gpio = new().  Should I redo that like #193 to support the lazy loading model being used everywhere else?  I can create a new issue for that.

josesimoes, 2022-08-03:
@jacobguilbeau I see that you've already fixed it. Looks good. Thank you!