nanoframework/Home #1114: [nanoFramework.Json] PopulateObject method suppress exceptions.

torbacz, 2022-08-06:
### Library/API/IoT binding

nanoFramework.Json

### Visual Studio version

VS2022

### Description

PopulateObject method in JsonConvert class is almost whole wrapped in try/catch. When error occurs during deserialization, it is just returning null value.

It may be problematic during runtime, instead of receiving "error when trying to desterilize" the code will return "NullReferenceException" from other method which is trying to use returned object (assuming it can't be null). 
Code example:
```csharp
var jsonObject = (TestObject)JsonConvert.DeserializeObject(jsonString, typeof(TestObject)); // The exception should occur here
var dataFromJson = jsonObject.Data; // The exception is raised here - NullReferenceException
```

More complex situation, when the API can return empty JSON (let's assume someone is crazy enough to do it :D) and there is some kind of null behavior the code will go into wrong execution path instead of throwing error.
Code example:
```csharp
var jsonObject = (TestObject)JsonConvert.DeserializeObject(jsonString, typeof(TestObject)); // The exception should occur here
if (jsonObject == null) // The code goes here and is returning 1 from method (this is valid, when API return null somehow)
 return 1;

return jsonObject.Data; // In fact it should return data property from object, but due to error inside of DeserializeObject method we received null.
```

More real world scenario is when we are trying to desterilize object A but the API returns object B, because of wrong API endpoint URL.

If you agree with me, I can submit PR :)

josesimoes, 2022-08-07:
I totally see what you're describing. And agreed that it could be improved. 
If you have a good idea on how to improve it, let's have that PR please! 

PS: probably better to wait for the refactor to be merged, right?

torbacz, 2022-08-07:
Yes, first refactor then new changes.

torbacz, 2022-08-08:
@josesimoes One more issue, tests are throwing exceptions, yet passing in runner (due to suppression of exceptions)
[image]

josesimoes, 2022-08-08:
Noted!

closed 2022-08-16 by josesimoes