nanoframework/Home #1115: [nanoFramework.Json] Unable to deserialize property when property name case doesn't match

torbacz, 2022-08-06:
### Library/API/IoT binding

nanoFramework.Json

### Description

With current implementation of JSON serialization the library is checking for properties when deserializing with exact match case which is not the default behavior in .net

https://docs.microsoft.com/en-us/dotnet/standard/serialization/system-text-json-character-casing
"The [web default](https://docs.microsoft.com/en-us/dotnet/standard/serialization/system-text-json-configure-options#web-defaults-for-jsonserializeroptions) is case-insensitive."

Solution:
Create new static class with configuration options for json serialization with one new property PropertyNameCaseInsensitive. When deserializing use new flag to determinate either use case sensitive names or not.

josesimoes, 2022-08-07:
This has already come up a few times. 
I'm a bit concerned on the performance impact that this will introduce... Code size will grow too, for sure.
Maybe we should measure all this to have a better understanding on the impact.

torbacz, 2022-08-07:
Well, I thinks it's just one setting - when comparing string you can specify if you want to ignore case or not.

```csharp
string.Equals(val, "astringvalue", StringComparison.OrdinalIgnoreCase)
```
or
```csharp
string.Equals(val, "astringvalue", StringComparison.Ordinal)
```
If it's implemented in nanoFrameowrk.

For sure, we should run some benchmarks before/after changes.

I've one more idea how to address this issue, but it's more complex and requires a lot of work. It will resolve some other issues also. I'll explain that in another issue.

josesimoes, 2022-08-07:
StringComparison is NOT available in nanoFramework.
We're using github issues to track workloads and confirmed bugs. Let's not use this for discussions. Either start a discussion or better use Discord, which allows a more streamlined conversation. 😉

closed 2022-10-20 by josesimoes