nanoframework/Home #1131: Support code secrets

networkfusion, 2022-08-23:
### Description

Given the use of nanoFramework in an IoT world, and its open source nature, it is likely that people will need to keep secrets and is easy to upload them by mistake.

nF should be able to support secrets more easily, and if possible (probably not) warn that they are about to be uploaded...

### How to solve the problem

A solution similar to https://dotnetcoretutorials.com/2022/04/28/using-user-secrets-configuration-in-net/

### Describe alternatives you've considered

Using template classes, and a nice bit of documentation to match...

derskythe, 2024-02-16:
To store WiFi secrets you can use [this solution](https://sandervandevelde.wordpress.com/2022/12/15/fun-with-nanoframework-running-net-charp-on-esp32/)

In few words, open `Device Explorer` windows and push the `Edit Network Configuration`, then tab `Wi-Fi profiles` and enter SID and Password.

Now, in code you can call secrets:

```CSharp
using System;
using System.Diagnostics;
using System.Threading;
 
using nanoFramework.Networking;
 
namespace ConnectToWIFITest
{
    public class Program
    {
        public static void Main()
        {
            ConnectToWifi();
 
            Thread.Sleep(Timeout.Infinite);
        }
        public static bool ConnectToWifi()
        {
 
            // Connect the ESP32 Device to the Wi-Fi and check the connection...
            Debug.WriteLine("Program Started, connecting to WiFi.");
 
            CancellationTokenSource cs = new(60000);
            var success = WifiNetworkHelper.Reconnect(requiresDateTime: true, token: cs.Token);
 
            if (!success)
            {
                Debug.WriteLine($"Can't connect to wifi: {WifiNetworkHelper.Status}");
                if (WifiNetworkHelper.HelperException != null)
                {
                    Debug.WriteLine($"NetworkHelper.ConnectionError.Exception");
                }
            }
 
            Debug.WriteLine($"Date and time is now {DateTime.UtcNow}");
            return success;
        }
    }
}
```

networkfusion, 2024-03-19:
@derskythe Just FYI, this issue relates to IDE secrets not being uploaded to a git repo, rather than secrets on a device.

Ellerbach, 2024-03-20:
@networkfusion I would argue that warning the user is not the job of nanoFramework extension. I would rather recommend the use of credentials scanners that are available as Visual Studio extensions (there are plenty of those). I would not reinvent something that does exist and is already availble for a lot of different secrets.