nanoframework/Home #1136: Regular expressions invalid display value in groups.

torbacz, 2022-08-27:
### Library/API/IoT binding

System.Text.RegularExpressions

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.2.023

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.495

### Description

Looks like there is something wrong with displaying System.Text.RegularExpressions.Group.Value in debug.

matches.Groups[2].Value -> displays in tooltip and Immediate Window "+CSQ: 0,0"
var value = matches.Groups[2].Value -> value has "0,0" value which is correct

Note: response = "AT+CSQ\r\n+CSQ: 0,0\r\n\r\nOK\r\n"

[image]

### Expected behaviour

The display value in debug should be the same as when assigned to variable.

Ellerbach, 2022-08-30:
@torbacz as the final result is correct, I would leave it like this for the moment. Not sure from where the issue is coming as the same flow is basically used in both cases. See: https://github.com/nanoframework/System.Text.RegularExpressions/blob/2c2043857777487019069ed7721e8c208c4867d1/nanoFramework.System.Text.RegularExpressions/Collections/GroupCollection.cs#L114

josesimoes, 2022-08-30:
This is most likely coming from the VS extension, in particular the way the inspect data is being processed and displayed.
Note that there are there somewhat convoluted escaping sequences and that is probably messing up with the processing.