nanoframework/Home #1137: nanoff ESP32 firmware update issue

Bengkel, 2022-09-01:
### Tool

nanoff

### Description

When flashing an ESP32 device with the latest version off nanoff, you will receive a warning: 

```
Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=16MB in order to remove this warning, or use the --dont-append-digest option
```

Although the flash procedure finishes with an OK, it failed. The device is not recognized by VS a/o nanoff as an nanoFramework device.

### How to reproduce

1. Update _Visual Studio_, the _nanoFramework extension_ and the _nanoff_ tool to the versions mentioned in this issue.
2. Open your command line console
3. Execute ```nanoff --serialport COM6 --update```
4. Execute ```nanoff --listdevices```: which will return ```No devices found```
5. Open _Visual Studio_ and navigate to the _device explorer extension_: the device will not be displayed

### Expected behaviour

After flashing the device, _nanoff_ and _Visual Studio_ should detect the device when it is connected to the computer.

### Screenshots

[image]
[image]
[image]

### Aditional context

Visual Studio 2022 Version 17.3.3
nanoFramework Extension version 2022.2.0.36
nanoff version 2.3.4
target  ESP32_REV0
firmware version ESP32_REV0-1.8.0.545

Capability output:
```
ESP32 (ESP32-D0WD (revision 1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 16MB unknown from ST (manufacturer 0x32 device 0x16408)
PSRAM: not available
Crystal 40MHz
MAC 0C:DC:7E:8E:1F:AC
```

josesimoes, 2022-09-01:
What you're seeing there it's a warning, it's not suposed to interfere with the flash writting proces, which the progress message seems to confirm.

1. what is the device that you have there?
2. Can you please connect a serial terminal to COM6, set the baudrate to 921600, reset the ESP32 and paste here the output you get from there?

Bengkel, 2022-09-01:
1. We use a [DFRobot FireBeetle](https://www.dfrobot.com/product-1590.html?search=wifi%20io&category_id=48&gclid=CjwKCAjwsMGYBhAEEiwAGUXJaSsa9t1iyHogPlMAwB3--WcO0vGNfP0CYEcou3Ij8yRuSOo61j7l3RoCKRgQAvD_BwE)
2a. Serial output [putty-nanoff-2.1.16.log](https://github.com/nanoframework/Home/files/9472533/putty-nanoff-2.1.16.log) when I reflash my device using nanoff 2.1.16
2b. Serial output [putty-nanoff-2.3.4.log](https://github.com/nanoframework/Home/files/9472550/putty-nanoff-2.3.4.log) when I reflash my device using nanoff 2.3.4

KiwiBryn, 2022-09-02:
I'm seeing similar but with a Sparkfun Thing Plus & RAK2305 . I though I had killed RAK2305 HW with FTDI adaptor fail but backing out nanoff versions it seems to be working again.
[image]
[image]
[image]
[image]

josesimoes, 2022-09-03:
@KiwiBryn you're using an veeeeery old version of nanoff! You really should update it.
There's a warning about this in yellow right at the top. Maybe we should find a way to highlight this if it's getting unnoticed... 🤔

KiwiBryn, 2022-09-03:
Hi @josesimoes I normally use the "latest & greatest"(currently 2.3.4) nanoff.

I was having problems with VS2K22 not discovering my ESP32(s) so I tried what @Bengkel suggested which required reverting back to 2.1.16 which appeared to work.  Have returned to current

Something odd going on with kernel panics on ESP32 when I use WatchChar need to get reliable repro rather than highjacking this thread

josesimoes, 2022-09-03:
Copy that @KiwiBryn !
Perfectly clear now. Was afraid that you could be fighting against the "wrong" kind of problems. 😜

closed 2022-09-05 by josesimoes