nanoframework/Home #1138: Build fails in VS 2022

profix898, 2022-09-01:
### Tool

Visual Studio extension

### Description

Trying to build any nanoFramework project with Visual Studio 2022 Community/Pro fails with the following log message:

`0>C:\Program Files\Microsoft Visual Studio\2022\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(729,5): Error MSB4062: The "GenerateNanoResourceTask" task could not be loaded from the assembly C:\Program Files\Microsoft Visual Studio\2022\MSBuild\nanoFramework\v1.0\nanoFramework.Tools.BuildTasks.dll. Could not load file or assembly 'file:///C:\Program Files\Microsoft Visual Studio\2022\MSBuild\nanoFramework\v1.0\nanoFramework.Tools.BuildTasks.dll' or one of its dependencies. An attempt was made to load a program with an incorrect format. Confirm that the <UsingTask> declaration is correct, that the assembly and all its dependencies are available, and that the task contains a public class that implements Microsoft.Build.Framework.ITask.`

I can confirm that the nanoFramework.Tools.BuildTasks.dll exists in that folder, but obviously it cant be loaded.

Any extra tools/components which must be installed in VisualStudio for the extension to work? Perhaps an older .NET Framework version (currently using VS with .NET FW 4.8.04084). I found some references (https://stackoverflow.com/questions/48294303/could-not-load-file-or-assembly-microsoft-build-utilities-core) that .NET x.y development tools might be required. I also found that the file in question (nanoFramework.Tools.BuildTasks.dll) references MSBuild v15.0 tools internally (which would be .NET Core 2.1 era and which are currently not present on the system).

### Aditional context

Visual Studio 2022 (17.3)
nanoFramework Extension 2022.2.0.37

josesimoes, 2022-09-02:
You probably need to update your VS to the latest (currently 17.3.2). 
We've been keeping our referenced NuGets close to the latest VS version being released.

Regarding reference to MSBuild v15.0 is not on our extension, it's coming from [Microsoft.Build.Framework](https://www.nuget.org/packages/Microsoft.Build.Framework). This is referencing that for quite sometime, IIRC, so all should be OK on that matter.

SpazCode153, 2022-09-04:
I have this same issue and I have the latest VS version which now is 17.3.3

josesimoes, 2022-09-05:
Just updated VS to 17.3.3 and I can't reproduce this. Tried with a couple of projects and everything is building just fine.
Can any of you @profix898 @SpazCode153 put here a link to a github repo with a solution that's failing so it can be investigated?

profix898, 2022-09-05:
I dont think its about the project. Even a freshly created (blank) project does not build. Nor do any of the samples from the official 'Samples' repo.

Its probably more a VS / tooling issue. I'm going to post my installed components as soon as I'm back on my dev box.

Sidenote: I noticed that on a second machine the nanoFramework templates didnt show up in the new project dialog. On that machine I dont have .Net 3.5 tools (for the .Net Framework 3.5 and 4.x projects) installed. That's why I still suspect missing VS components or missing dependencies.

Does the nanoFramework extension write a log file somewhere?

josesimoes, 2022-09-05:
So far there are only reports of this from you two, so I'm strongly suspecting this is a local problem.

About the log, that is generated by VS on extension install. It shows the location of it in the last install step. 

Make sure that you install the workloads for .NET desktop development and .NET Core cross-platform development.

SpazCode153, 2022-09-06:
Okay I think my problem might be slightly different. I get the error ".NET nanoFramework GenerateNanoResourceTask error: Exception of type 'System.Exception' was thrown." only when I build the samples to do with graphics like the Tetris sample or SimpleWpf.

I tried to look at the build output with verbose logging enabled but I cant find the cause of the problem there, it still only tells me that the GenerateNanoResourceTask failed and threw 'System.Exception'.

Is there anywhere else that I can look at logs for a possible cause to the problem?

josesimoes, 2022-09-06:
@SpazCode153 there's a known issue with compiling some types of embedded resources. That's probably what you're seeing.

SpazCode153, 2022-09-06:
Oh I see, sorry.

Is there any way that you know of to overcome it? Perhaps things to try?

josesimoes, 2022-09-06:
> Is there any way that you know of to overcome it? Perhaps things to try?
The problem is identified and will be fixed. Not much to try and no way to overcome except working on the fix. 😁

SpazCode153, 2022-09-06:
Not a problem. Thanks @josesimoes.

profix898, 2022-09-06:
I just updated VS to the latest preview (17.4 preview 1) and the issue is gone. I can now build a newly created nanoFramework project without any issues. The nominal list of installed components hasn't changed. Very strange. Anyway, I think the issue can be closed. Thanks for your help.

closed 2022-09-06 by profix898