nanoframework/Home #1142: nanodu is failing to submit PR against 3rd party repos

josesimoes, 2022-09-14:
### Tool

nanodu

### Description

The tool is failing to submit the PR with the updates against a 3rd party repo.

### How to reproduce

See log from this github action run: https://github.com/nanoframework/amqpnetlite/actions/runs/3054202425

josesimoes, 2022-09-14:
@torbacz any ideas on what could be happening?

josesimoes, 2022-09-14:
That's clearly something with the fact that's a fork from "upstream". I did add the repoOwner parameter and that's being passed correctly, so it has to be something with the API call that creating the PR...

torbacz, 2022-09-14:
@josesimoes 
Just checked log again. 
[image]

Shouldn't it be "nanoframework"?

josesimoes, 2022-09-14:
Nope. Because the PR is meant to be submitted against the upstream repo.