nanoframework/Home #1144: Bluetooth library does not expose the Generic Access and Appearance Characteristic

Ellerbach, 2022-09-18:
### Library/API/IoT binding

nanoFramework.Device.Bluetooth

### Description

The Generic Access service (0x1800) is created when the StartAdvertising function is called.
But, you cannot adjust it, you can't add a second one as only the one used for the advertising will be used.
It should be exposed like any other service and even be created since the beginning like a service always present. 
So something like GattServiceProvider.Create() (without anything in there), should just return this basic and always present element.
Some applications require to access the Appearance characteristic and change the value (example: all HID devices).
Other option would be to have the ability to have it passed in the parameters of the StartAdvertising function.

### Expected behaviour

Being able to access the Appearance Characteristic from the Generic Access service.

AdrianSoundy, 2023-02-26:
Currently this is being added automatically by the stack.

Will look at adding this the first time a service is created so it can be updated by user.

Ellerbach, 2023-03-01:
> Will look at adding this the first time a service is created so it can be updated by user.

Yes, this will be great as you have to be able to adjust it. Otherwise we can't do HID for example because you have to pass additional elements. So it's important to let the user pass it.

AdrianSoundy, 2023-03-11:
The generic access contains the device name which is current set and the appearance.
The appearance will be added as a property to the BluetoothLEServer class and default to "Generic unknown device".  The appearance is also optional in the advertisment packet but will leave that until next PR when adverticments are improved.

AdrianSoundy, 2023-03-13:
Changes made to server so appearance is now set in generic access service. Also changed client so the name and appearance is available in BluetoothLEDevice.

Currently testing changes and updating samples.

closed 2023-03-16 by AdrianSoundy