nanoframework/Home #1147: Deploy button is inaccessible after update to VS Extension v2022.2.0.41

oloendithas, 2022-09-21:
### Tool

Visual Studio extension

### Description

Deploy button is inaccessible (grayed out) after update to VS Extension v2022.2.0.41.
Rolling back to v2022.2.0.39 solved the problem.

VS Community 2022 v17.3.4
Board ESP32 ROM 4mb, no PSRAM
Projects: any, including official examples.

### How to reproduce

Update VS extension to v2022.2.0.41
Try to press "Deploy project" button

### Expected behaviour

Deploy started

josesimoes, 2022-09-21:
Where is this button? Or are you referring to the various menus where the "Deploy..." command shows up?

BTW: know that you don't have to _deploy_ a project. When you start a debug session VS will take care of deploying everything for you.

oloendithas, 2022-09-21:
I mean this button
[image]

josesimoes, 2022-09-21:
OK that's a menu item.

In any case, that won't be enabled if there is something wrong with the project system not loading correctly or if the nanoFramework extension is not loaded.

I'm on same version as you and I don't have any issues with that.
Maybe this is related with #1009. Try to execute the command as explained there.

If this does not solve your issue, we're better continuing on Discord for a more streamlined chat.