nanoframework/Home #1152: RegularExpressions:  Group Captures collection has only the last capture. Unit test RegExpTest_6_Group_Test_0 gives false positive.

jazzman55, 2022-09-23:
### Library/API/IoT binding

System.Text.RegularExpressions

### Visual Studio version

VS2022

### Description

When regular expression match group captures multiple items, only the last one appears in Group[_GROUP_NO_].Captures.
This is not caught by unit tests, as GroupTests.RegExpTest_6_Group_Test_0 gives false positive. See screenshot.

### How to reproduce

1. Run RegExpTest_6_Group_Test_0 test.
2. Observe output and compare with comments in the test.

### Expected behaviour

- Output should match the comment i.e. multiple captures appear in groups 2 and 3.
- Test should fail if captures count does not match expected numbers.

### Screenshots

[image]

torbacz, 2022-09-23:
Related to https://github.com/nanoframework/Home/issues/1136?

Ellerbach, 2022-09-23:
@jazzman55, it's just a debug related issue as @torbacz pointed out from my perspective. That will require quite some investigation.
Have you tried to run the code and release mode?

jazzman55, 2022-09-23:
It doesn't work in Release either. I added a few Asserts that validate Captures counts and they fail in both Debug and Release. See screenshot.
[image]

Ellerbach, 2022-09-23:
ok, so definitely a bug. If you have time to dig into, we're always happy to get PR to fix those :-)