nanoframework/Home #1156: Static field, called from inside static field of `Program` class not initialized

cbagpipe, 2022-09-27:
### Library/API/IoT binding

BCL

### Visual Studio version

VS2022 (17.3.4)

### .NET nanoFramework extension version

2022.2.0.41

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.594

### Device capabilities

```text
System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 1b16ef6
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Sep 22 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.594
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Teleavtomatika.Nanoframework.SampleApp, 1.0.0.0
  System.IO.FileSystem, 1.1.5.25011
  nanoFramework.Logging.Serial, 1.1.19.269
  System.Threading, 1.1.8.6695
  nanoFramework.Logging, 1.1.19.269
  nanoFramework.System.Text, 1.2.7.60109
  System.Device.Wifi, 1.5.22.4970
  mscorlib, 1.12.0.4
  System.Math, 1.4.4.3
  System.IO.Streams, 1.1.15.36091
  System.Net, 1.10.21.18203
  System.Net.Sockets.TcpClient, 1.1.17.41463
  System.Net.Sockets.UdpClient, 1.1.16.46254
  nanoFramework.System.Collections, 1.4.0.3
  nanoFramework.Runtime.Events, 1.10.0.3
  System.IO.Ports, 1.1.25.11551
  nanoFramework.Runtime.Native, 1.5.4.3
  System.Device.Gpio, 1.1.18.20780
  Teleavtomatika.NanoFramework.Core, 1.0.0.0
  Windows.Storage, 1.5.6.13278
  nanoFramework.Json, 2.2.29.62400
  Windows.Storage.Streams, 1.14.8.30993

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty
```

### Description

Static field contained inside class, initialized in static field of `Program` class, not populated with value.

### How to reproduce

1. Create empty project
2. Paste example code below
3. Set breakpoint after `gettingNull` variable
4. Run debug

### Expected behaviour

variable `gettingNull` MUST be equal to static field `test`

### Screenshots

[image]

### Sample project or code

```csharp
public class Program
    {
        static TestClass testInstance = new TestClass();

        public static void Main()
        {
            Thread.Sleep(Timeout.Infinite);
        }
    }

    public class TestClass
    {
        public static string test = "abc";

        public TestClass()
        {
            var gettingNull = test;
        }
    }
```

### Aditional information

I don't know where to address this bug, so addressing to BCL. If I mistaken, please, fix this issue description.

klausvancamelbeke, 2022-10-04:
Seeing the same issue on ESP_WROVER_KIT 1.8.0.534 :
[image]

I debugged while taking System.IO.FileSystem into my solution and the static field InvalidPathChars remains null.
https://github.com/nanoframework/System.IO.FileSystem/blob/0ccac9e0ecf10844f64ad90b942407dcba47c439/System.IO.FileSystem/Path.cs#L38

cbagpipe, 2023-06-06:
Suffering from this bug again.
I have static `Helpers` class with static `Crc` class in it with static property. So, this property remains null and debugger no stopping inside static constructor.
Workaround → made class a singleton

josesimoes, 2023-06-06:
This static initialization can be very complex. That's the reason why this issue is still open...
I've start looked into it and it requires a serious amount of hours to get to the bottom of it.

josesimoes, 2026-11-02:
This has been fixed in nanoframework/nf-interpreter#3589. 
Note that this is the develop branch so fix is available only in preview fw versions.

closed 2026-11-02 by josesimoes