nanoframework/Home #1162: Add packages lock to all libraries

josesimoes, 2022-10-03:
### Details about Problem

Need to add packages lock file to all class library repos.

### Description

Use this PR as guideline: nanoframework/nanoFramework.Networking.Sntp#118.

Ellerbach, 2022-10-04:
Maybe you can as well give the steps on how to generate/create the packages lock file?

networkfusion, 2022-10-04:
> Maybe you can as well give the steps on how to generate/create the packages lock file?

As simple as building the project in visual studio after adding an empty file.

networkfusion, 2022-10-08:
Adding this for reference... https://github.com/search?q=org%3Ananoframework+project.lock&type=code
Also including the warning about `gitignore` excluding `project.lock.json` in some repos!

josesimoes, 2022-11-18:
This should be pretty much completed. Any library that's missing it will be updated later.

closed 2022-11-18 by josesimoes