nanoframework/Home #1163: Update github actions on all repos

josesimoes, 2022-10-03:
### Details about Problem

Update/add github actions to most of the repos.

### Description

Use this as inspiration: https://github.com/nanoframework/nanoFramework.Networking.Sntp/tree/main/.github/workflows

weslleymurdock, 2022-10-08:
Hello, I'm interested to help on this one. How i may help here?

networkfusion, 2022-10-08:
Hey @weslleymurdock , this is interlinked with  #1162 , please check there and choose a repo that needs work (basically any of the managed libs that have not been updated in the last couple of weeks), and make the suggested changes.

weslleymurdock, 2022-10-08:
Thanks a lot. I will take a look on the mentioned issue and start

networkfusion, 2023-01-15:
@josesimoes I think we are probably done here?!

josesimoes, 2023-01-16:
I would agree.

closed 2023-01-16 by josesimoes