nanoframework/Home #1164: Typeof for array types returns invalid data

torbacz, 2022-10-03:
### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.629

### Was working before? On which version?

Yes, No idea about version.

### Description

Not quite sure what's going on. One time I've got "string" as type. It was working previously, because benchmarks for JSON project has int[] and short[] deserialization test.

Also, based on nanoFramework.Json.JsonConvert.PopulateObject typeof(int[]).GetElementType() should return "int[]" type.

### How to reproduce

var type = typeof(short[]);
Console.WriteLine(type.ToString());
Console.WriteLine(type.Name);
Console.WriteLine(type.GetElementType().FullName);

### Screenshots

[image]

josesimoes, 2022-10-04:
Most likely related with a recent fix for array's type...
This means that mscorlib Unit Tests are missing tests for these, otherwise it would have been caught there.

Ellerbach, 2023-04-18:
And to help a bit, this is actually working:

```csharp
ushort[] array = new ushort[10];

  var type = array.GetType();
  Console.WriteLine(type.ToString());
  Console.WriteLine(type.Name);
  Console.WriteLine(type.GetElementType().FullName);
```

That does return properly:

```text
System.UInt16[]
UInt16[]
System.UInt16
```

so issue seems to be with `typeof` for the array system types.

Ellerbach, 2023-04-18:
And this crashes directly: `var type = typeof(System.UInt16[]);`

closed 2023-06-02 by josesimoes