nanoframework/Home #1166: OLIMEX has no wifi support

thombrink, 2022-10-07:
### Library/API/IoT binding

System.Device.Wifi

### Description

When running ESP32_OLIMEX the 'ScanAsync' method throws an InvalidOperationException exception:

```
    ++++ Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ++++
    ++++ Message: 
    ++++ System.Device.Wifi.WifiAdapter::NativeScanAsync [IP: 0000] ++++
    ++++ System.Device.Wifi.WifiAdapter::ScanAsync [IP: 000d] ++++
```

#951 does not solve the problem after implementing the 'CheckWifiInterface' interface from discord and using the latest nugets and firmware.

josesimoes, 2022-10-07:
OLIMEX does have Wifi support, otherwise you wouldn't even been able to deploy the application on it.
Please check the [WiFi sample](https://github.com/nanoframework/Samples/tree/main/samples/Wifi) about using scan.

You're better asking this in our Discord channel then here. We're using github issues to track work and give visibility to confirmed issues. Not to provide support or answer questions. 😉

thombrink, 2022-10-07:
Sure!

Created a thread here: https://discord.com/channels/478725473862549535/481782754674212867/1027966089117847552

Feiko, 2022-10-14:
I'm experiencing the same issue with the M5Stack Atom Lite (ESP32-PICO platform). Tried the default WiFi sample and played around with the network configuration settings in the VS extension. However, I get the same exception when using the ScanAsync method.

JORGEGO, 2022-10-27:
Same error!!! 3 weeks stopper not connect

Ellerbach, 2022-10-27:
I've started some investigation and regarding the dates and what is impacted, it seems possible that this PR has adjusted something: https://github.com/nanoframework/nf-interpreter/pull/2365
More precisely, looking at the changes, it can come from here:
https://github.com/nanoframework/nf-interpreter/pull/2365/files#diff-97ea37fc6cd18b4e9204849a7619d0def3ee8c616f04d63714af8208c0de9a12

@AdrianSoundy would you have an idea of the problem? Can this be linked to this change you've been doing?

josesimoes, 2022-10-27:
Confirmed with ScanWifi sample running on ATOM. 
Not working with latest fw and nugets. Working with fw 1.8.0.534 and Sys.Dev.Wifi 1.5.8.

AdrianSoundy, 2022-10-28:
I'll have a look to see what's going on.

JORGEGO, 2022-10-28:
Thanks @AdrianSoundy it is important for us.

zamby, 2022-10-29:
I have same problem with ESP32 WEMOS.

Il thread '<Nessun nome>' (0x2) è terminato con il codice 0 (0x0).
starting Wifi scan
    ++++ Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ++++
    ++++ Message: 
    ++++ System.Device.Wifi.WifiAdapter::NativeScanAsync [IP: 0000] ++++
    ++++ System.Device.Wifi.WifiAdapter::ScanAsync [IP: 0010] ++++
Eccezione generata: 'System.InvalidOperationException' in System.Device.Wifi.dll
message:Exception was thrown: System.InvalidOperationException
stack:System.Device.Wifi.WifiAdapter::NativeScanAsync
System.Device.Wifi.WifiAdapter::ScanAsync

AdrianSoundy, 2022-10-31:
The issue is being caused by an out of native memory when it tries to init the Wifi stack.   Reducing amount of memory used for managed heap by 10K resolves problem but that's probably not the root cause. Something else has changed. Will investigate further.

AdrianSoundy, 2022-10-31:
For PICO builds problem caused by adding SPIRAM to the Pico builds. As rev1 chip reduces native heap by at least 10K when it adds alternative  C library with spiram fixes.  So 1.8.0.594 last good build ( Builds 611 & 629 broken) 642 has spiram enabled.

alberk8, 2022-10-31:
> For PICO builds problem caused by adding SPIRAM to the Pico builds. As rev1 chip reduces native heap by at least 10K when it adds alternative C library with spiram fixes. So 1.8.0.594 last good build ( Builds 611 & 629 broken) 642 has spiram enabled.

We should just disable SPIRAM for pico.  The M5Stack Atom series does not have psram and this will be the most used ESP32-PICO devices.

josesimoes, 2022-10-31:
Or, because we can, let's leave PSRAM enabled for this build and compensate those 10k. Doesn't seem much loss anyway...

AdrianSoundy, 2022-10-31:
Ok i'll remove 10K from managed heap which works for Olimex and Pico boards
I checked that the Pico rev 1 can have SPI ram.

josesimoes, 2022-10-31:
Yes please!

JORGEGO, 2022-11-01:
Excellent as soon as will be ready the version I will test. Thanks for your great effort

JORGEGO, 2022-11-02:
It seems still not work... Now in my case after upgrading the last firmware and the las version assemblies, exception it is not producing, but not detect any Wi-Fi network
starting Wifi scan
Wifi_AvailableNetworksChanged - get report
starting Wifi scan
Wifi_AvailableNetworksChanged - get report
starting Wifi scan
Wifi_AvailableNetworksChanged - get report
starting Wifi scan
Wifi_AvailableNetworksChanged - get report

JORGEGO, 2022-11-03:
Nothing... with the new firmware and package starting to crash other assemblies
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
    ++++ Exception System.Exception - CLR_E_FILE_IO (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.FileIO::ReadTextNative [IP: 0000] ++++
    ++++ Windows.Storage.FileIO::ReadText [IP: 0008] ++++
    ++++ greenyng.device.nanoframework.atomlite.scale001.Persistence::LoadContext [IP: 000a] ++++
    ++++ greenyng.device.nanoframework.atomlite.scale001.Program::Main [IP: 0030] ++++
Exception thrown: 'System.Exception' in Windows.Storage.dll
    ++++ Exception System.ArgumentException - 0xfd000000 (1) ++++
    ++++ Message: 
    ++++ System.IO.Ports.SerialPort::NativeInit [IP: 0000] ++++
    ++++ System.IO.Ports.SerialPort::Open [IP: 000f] ++++
    ++++ greenyng.device.nanoframework.atomlite.scale001.qr.QRUtilities::InitQRCode [IP: 001e] ++++
    ++++ greenyng.device.nanoframework.atomlite.scale001.Navigation.Manufacturer::ReadIdDevice [IP: 0009] ++++
    ++++ greenyng.device.nanoframework.atomlite.scale001.Program::Main [IP: 0030] ++++
Exception thrown: 'System.ArgumentException' in System.IO.Ports.dll
An unhandled exception of type 'System.ArgumentException' occurred in System.IO.Ports.dll

JORGEGO, 2022-11-03:
How could I get the 1.8.0.534 firmware?

JORGEGO, 2022-11-03:
I could not lost more time...

JORGEGO, 2022-11-03:
nanoff --target ESP32_PICO --fwversion 1.8.0.534 --update --serialport COM6

josesimoes, 2022-11-15:
The "reporting" on the cause of the failure has been improved.

@thombrink most likely what the cause for the failure was that the scan operation was being attempted when the station was still in the process of connecting. This is one of the most common causes for scan to fail and our API wasn't reporting that conveniently.

closed 2022-12-11 by josesimoes

zhulige, 2023-03-06:
ESP32_LILYGO Same problem

zhulige, 2023-03-06:
ets Jul 29 2019 12:21:46[2023-03-06 04:54:47.547]
[2023-03-06 04:54:47.547]
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)[2023-03-06 04:54:47.563]
configsip: 0, SPIWP:0xee[2023-03-06 04:54:47.563]
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00[2023-03-06 04:54:47.563]
mode:DIO, clock div:2[2023-03-06 04:54:47.563]
load:0x3fff0030,len:5688[2023-03-06 04:54:47.563]
load:0x40078000,len:14528[2023-03-06 04:54:47.563]
load:0x40080400,len:3708[2023-03-06 04:54:47.563]
entry 0x4008067c[2023-03-06 04:54:47.578]
[0;32mI (27) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v4.4.4[0m[2023-03-06 04:54:47.578]
[0;32mI (27) boot: build Feb 27 2023 15:13:27[0m[2023-03-06 04:54:47.594]
[0;32mI (29) boot: chip revision: v3.0[0m[2023-03-06 04:54:47.594]
[0;32mI (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m[2023-03-06 04:54:47.594]
[0;32mI (40) boot.esp32: SPI Speed      : 40MHz[0m[2023-03-06 04:54:47.610]
[0;32mI (44) boot.esp32: SPI Mode       : DIO[0m[2023-03-06 04:54:47.610]
[0;32mI (49) boot.esp32: SPI Flash Size : 4MB[0m[2023-03-06 04:54:47.610]
[0;32mI (53) boot: Enabling RNG early entropy source...[0m[2023-03-06 04:54:47.610]
[0;32mI (59) boot: Partition Table:[0m[2023-03-06 04:54:47.626]
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m[2023-03-06 04:54:47.626]
[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m[2023-03-06 04:54:47.626]
[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m[2023-03-06 04:54:47.642]
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 001a0000[0m[2023-03-06 04:54:47.658]
[0;32mI (92) boot:  3 deploy           Unknown data     01 84 001b0000 001f0000[0m[2023-03-06 04:54:47.658]
[0;32mI (99) boot:  4 config           Unknown data     01 82 003c0000 00040000[0m[2023-03-06 04:54:47.658]
[0;32mI (107) boot: End of partition table[0m[2023-03-06 04:54:47.673]
[0;32mI (111) boot_comm: chip revision: 3, min. application chip revision: 0[0m[2023-03-06 04:54:47.673]
[0;32mI (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2b1f8h (176632) map[0m[2023-03-06 04:54:47.689]
[0;32mI (190) esp_image: segment 1: paddr=0003b220 vaddr=3ff80067 size=00008h (     8) load[0m[2023-03-06 04:54:47.753]
[0;32mI (191) esp_image: segment 2: paddr=0003b230 vaddr=3ffb0000 size=03574h ( 13684) load[0m[2023-03-06 04:54:47.768]
[0;32mI (201) esp_image: segment 3: paddr=0003e7ac vaddr=40080000 size=0186ch (  6252) load[0m[2023-03-06 04:54:47.768]
[0;32mI (207) esp_image: segment 4: paddr=00040020 vaddr=400d0020 size=cd658h (841304) map[0m[2023-03-06 04:54:47.779]
[0;32mI (514) esp_image: segment 5: paddr=0010d680 vaddr=4008186c size=1a968h (108904) load[0m[2023-03-06 04:54:48.104]
[0;32mI (559) esp_image: segment 6: paddr=00127ff0 vaddr=400c0000 size=00068h (   104) load[0m[2023-03-06 04:54:48.123]
[0;32mI (573) boot: Loaded app from partition at offset 0x10000[0m[2023-03-06 04:54:48.136]
[0;32mI (573) boot: Disabling RNG early entropy source...[0m[2023-03-06 04:54:48.136]
[0;32mI (584) psram: This chip is ESP32-D0WD[0m[2023-03-06 04:54:48.136]
[0;31mE (585) psram: PSRAM ID read error: 0xffffffff[0m[2023-03-06 04:54:48.136]
[0;31mE (585) spiram: SPI RAM enabled but initialization failed. Bailing out.[0m[2023-03-06 04:54:48.154]
[0;32mI (592) cpu_start: Failed to init external RAM; continuing without it.[0m[2023-03-06 04:54:48.154]
[0;32mI (599) cpu_start: Pro cpu up.[0m[2023-03-06 04:54:48.154]
[0;32mI (603) cpu_start: Starting app cpu, entry point is 0x40085e10[0m[2023-03-06 04:54:48.167]
[0;32mI (0) cpu_start: App cpu up.[0m[2023-03-06 04:54:48.167]
[0;32mI (625) cpu_start: Pro cpu start user code[0m[2023-03-06 04:54:48.167]
[0;32mI (625) cpu_start: cpu freq: 240000000[0m[2023-03-06 04:54:48.199]
[0;32mI (625) cpu_start: Application information:[0m[2023-03-06 04:54:48.199]
[0;32mI (629) cpu_start: Project name:     nanoCLR[0m[2023-03-06 04:54:48.199]
[0;32mI (634) cpu_start: App version:      1.8.1.11[0m[2023-03-06 04:54:48.199]
[0;32mI (639) cpu_start: Compile time:     Feb 27 2023 15:12:59[0m[2023-03-06 04:54:48.217]
[0;32mI (645) cpu_start: ELF file SHA256:  198409da2436be8c...[0m[2023-03-06 04:54:48.217]
[0;32mI (651) cpu_start: ESP-IDF:          v4.4.4[0m[2023-03-06 04:54:48.217]
[0;32mI (657) heap_init: Initializing. RAM available for dynamic allocation:[0m[2023-03-06 04:54:48.231]
[0;32mI (664) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m[2023-03-06 04:54:48.231]
[0;32mI (670) heap_init: At 3FFBC660 len 000239A0 (142 KiB): DRAM[0m[2023-03-06 04:54:48.231]
[0;32mI (676) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m[2023-03-06 04:54:48.262]
[0;32mI (682) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m[2023-03-06 04:54:48.262]
[0;32mI (689) heap_init: At 4009C1D4 len 00003E2C (15 KiB): IRAM[0m[2023-03-06 04:54:48.262]
[0;32mI (696) spi_flash: detected chip: generic[0m[2023-03-06 04:54:48.262]
[0;32mI (700) spi_flash: flash io: dio[0m[2023-03-06 04:54:48.262]
[0;32mI (715) cpu_start: Starting scheduler on PRO CPU.[0m[2023-03-06 04:54:48.279]
[0;32mI (0) cpu_start:(D?D

zhulige, 2023-03-06:
starting Wi-Fi scan
    ++++ Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ++++
    ++++ Message: 
    ++++ System.Device.Wifi.WifiAdapter::NativeScanAsync [IP: 0000] ++++
    ++++ System.Device.Wifi.WifiAdapter::ScanAsync [IP: 000d] ++++

alberk8, 2023-03-06:
@zhulige ,  Load a firmware without PSRAM or use `nanoff --platform esp32 --update --serialport COM3 ` to load the firmware.

zhulige, 2023-03-06:
@alberk8 Ethernet cannot be used。

alberk8, 2023-03-06:
Then you need to do a manual compile if you want Ethernet with Wifi or get the exact same esp32 module that support both.

zhulige, 2023-03-07:
@alberk8 Ethernet + Wifi  Can I use it at the same time?

alberk8, 2023-03-07:
Have not tried, most likely you cannot.

MaUlBit, 2023-05-06:
> @zhulige , Load a firmware without PSRAM or use `nanoff --platform esp32 --update --serialport COM3 ` to load the firmware.

This solved my issue on the Arduini huzzah! ESP32-WROOM-32E
Thanks for the hint!