nanoframework/Home #1170: Direct Method doesn't work for PnP Device with components

romankiss, 2022-10-19:
### Library/API/IoT binding

nanoFramework.Azure.Devices.Client

### Visual Studio version

VS2022 Version 17.3.6

### .NET nanoFramework extension version

2022.2.0.44

### Target name(s)

M5StickCPlus

### Firmware version

1.8.0.643

### Description

Source file:
[nanoFramework.Azure.Devices](https://github.com/nanoframework/nanoFramework.Azure.Devices)/[nanoFramework.Azure.Devices.Client](https://github.com/nanoframework/nanoFramework.Azure.Devices/tree/main/nanoFramework.Azure.Devices.Client)/DeviceClient.cs

1. **DirectMethodTopic** (line 447+) is working well for default (root) component direct method, but for other ones will never invoke its delegate, in other words there is no match for methodName contains a star character, for instance, thermostat1*getMaxMinReport, therefore this part must be fixed replacing by underscore ('_') character and delegate name should be used as thermostat1_getMaxMinReport.

2. **DirectMethodTopic** Return Call (line 468) doesn't handle a long-running command
( https://learn.microsoft.com/en-us/azure/iot-central/core/howto-use-commands#long-running-commands) where a response back return 202 code, such as Accepted to indicate, that the result is pending. For this case the delegate will return either the _res=null_ or _res=""_ and based on this, the topic should be in the line 468 looks like the following: $"$iothub/methods/res/202/?$rid={rid:X}". Note, that the payload should be shows a text, that the command is in pending process.

Thanks
Roman

Ellerbach, 2022-10-20:
Thanks for the sumission. 

On case 1: you basically ask to replace in the coming "Topic" the character `*` by `_` so it can be used in the methog name. correct? If it's about this, please do not hesitate to provide a PR.

On case 2: correct, there is no support for long running. nanoFramework is nano, so there are couple of elements that we decided not to implement for simplification. If that's business critical for you, feel free to propose a PR to add this capability.

josesimoes, 2022-11-23:
@romankiss can you please point me to the Azure docs where the naming handling is explained? (the 1st part about replacing the * with _)

romankiss, 2022-11-23:
Hi @josesimoes , the naming handling for direct method invoked on the component in the PnP device is described in the [IoT Plug and Play device developer guide](https://learn.microsoft.com/en-us/azure/iot-develop/concepts-developer-guide-device?pivots=programming-language-ansi-c#commands) and [IoT Plug and Play conventions](https://learn.microsoft.com/en-us/azure/iot-develop/concepts-convention?source=recommendations#commands).

josesimoes, 2026-10-01:
We looked at this again against the current docs.

Component commands (thermostat1*getMaxMinReport): confirmed and still relevant. The IoT Plug and Play convention still sends component commands as componentName*commandName, and DeviceClient compared that whole string against the C# method name, so it could never match.
Instead of renaming callbacks with _ (DTDL names can contain _ themselves, which would make it ambiguous), we're adding an explicit overload:
`azureIoT.AddMethodCallback(getMaxMinReport, "thermostat1");`

Hope this satisfies the requirement.
See nanoframework/nanoFramework.Azure.Devices#471

josesimoes, 2026-10-01:
Regarding the long-running commands: those are an IoT Central convention, not part of IoT Plug and Play, and IoT Central is being retired. We won't add IoT Central-specific behaviour (such as "null -> 202"). Instead, callbacks can now return any status code through a new MethodResponseCallback/MethodResponse, which covers 202 and any other status IoT Hub accepts:
`MethodResponse RunDiagnostics(int rid, string payload) => new MethodResponse(202, "{\"status\":\"pending\"}");`

While doing this also fixed a bug with a callback returning null being reported as 504, and unknown methods now get a 501 reply instead of timing out.

Thank you again @romankiss for the detailed report and the doc pointers!

closed 2026-10-01 by josesimoes