nanoframework/Home #1171: Missing a topic argument in the SendMessage method.

romankiss, 2022-10-19:
### Library/API/IoT binding

nanoFramework.Azure.Devices.Client

### Visual Studio version

VS2022 Version 17.3.6

### .NET nanoFramework extension version

2022.2.0.44

### Target name(s)

M5StickCPlus

### Firmware version

1.8.0.643

### Description

source file:
[nanoFramework.Azure.Devices](https://github.com/nanoframework/nanoFramework.Azure.Devices)/[nanoFramework.Azure.Devices.Client](https://github.com/nanoframework/nanoFramework.Azure.Devices/tree/main/nanoFramework.Azure.Devices.Client)/DeviceClient.cs

**SendMessage** (line 371) is using hardcoded topic (_telemetryTopic) for publishing. There is no way to add some application properties on the published topic, for instance: $.ct=application%2Fjson&$.ce=utf-8
and others depended on application, also there is no way to send message on another topic for instances:
$iothub/methods/res/200/?$rid=123 or $iothub/streams/res/202/?$rid=2 etc.
So, the signature of SendMessage method should be looked like the following:

     bool SendMessage(string message, CancellationToken cancellationToken = default, string dtdlComponentname = "", string pubTopic = "")

if the **pubTopic = ""** or null, the topic will be used from the property __telemetryTopic_ otherwise from the method _pubTopic_.
It is very simple change, without any impact, but it will allow to make more useful things.

Thanks
Roman

Ellerbach, 2022-10-20:
Feel free to propose a PR for this one as well!

romankiss, 2022-10-20:
Hi @Ellerbach,

I don't have an experience with Github PR, please could you make the following changes (replacing lines):

source file: 
[nanoFramework.Azure.Devices](https://github.com/nanoframework/nanoFramework.Azure.Devices)/[nanoFramework.Azure.Devices.Client](https://github.com/nanoframework/nanoFramework.Azure.Devices/tree/main/nanoFramework.Azure.Devices.Client)/DeviceClient.cs

line 371:

    public bool SendMessage(string message, CancellationToken cancellationToken = default, string dtdlComponentname = "", string pubTopic = null)

line 373:

    string topic = pubTopic ?? _telemetryTopic;

line 377:

    topic += $"{(topic.EndsWith("/") ? "" : "&")}$.sub={dtdlComponentname}";

- also the case described in the issue 1170,  inserting two following lines after line 450 (replacing a star character in the methodName):

        int ii = methodName.IndexOf('*');
        methodName = ii > 0 ? $"{methodName.Substring(0, ii)}_{methodName.Substring(ii + 1)}" : methodName;

 I didn't see an extension class for string, so the replacement character is done by inline code.

Thanks
Roman

josesimoes, 2022-11-18:
This is called "property bag" in [Azure docs](https://learn.microsoft.com/azure/iot-hub/iot-hub-mqtt-support?WT.mc_id=WT.mc_id%3DDT-MVP-5004179#receiving-cloud-to-device-messages).

romankiss, 2022-11-23:
Hi @josesimoes & @Ellerbach, thanks for improving the _SendMessage_ method. 

Besides that, this improving doesn't handle posibility of the SendMessage method to publish it on the custom topic, for example, different response code (202) must be returned for long running command ( https://learn.microsoft.com/en-us/azure/iot-central/core/howto-use-commands#long-running-commands).

Thanks again
Roman

josesimoes, 2022-11-23:
Thanks for reporting back.

About the long run command that will be implemented in a future work batch.