nanoframework/Home #1172: JsonConvert incorrectly deserializes strings with non-ASCII characters

jazzman55, 2022-10-25:
### Library/API/IoT binding

nanoframework.Json

### Description

When an object contains string with non-ASCII characters (i.e. unicode chars encoded on two bytes), after deserializalization the string doesn't match the original.

### How to reproduce

1. Create an object with string property. Set some non-ASCII characters to the property (e.g. 'ł', '\u0165').
2. Serialize the object with JsonConvert.
3. Deserialize the object.

### Expected behaviour

Deserialized object's property value equals original object's property value.

### Sample project or code

[TestMethod]
public void SerializationWithNonAscii()
{
    var input = new TestClass { Prop1 = "zażółć gęślą jaźń" };
    var str = JsonConvert.SerializeObject(input);
    var result = (TestClass)JsonConvert.DeserializeObject(str, typeof(TestClass));
    Assert.Equal(input.Prop1, result.Prop1);
}

### Aditional information

In JsonConvert class, method GetNextTokenInternal() iterates through byte array and casts single byte to char:
ch = (char)jsonBytes[jsonPos++];
If character is stored on two bytes it is processed incorrectly as two separate characters. Possible fix would be to iterate on string instead of byte array:
ch = jsonString[jsonPos++];

jazzman55, 2022-10-25:
Hey @josesimoes !
Looks like currently it's just the ASCII subset of UTF-8. According to https://en.wikipedia.org/wiki/UTF-8 we should be able to encode over-one-byte chars as well. String class itself works fine with those, it's just the JSON library where I've encountered this issue. I see the value in enabling unicode support in JSON, as it opens up for interacting with APIs in all world's languages.
BTW, I have a fix based on the suggestion above, but it has performance penalty. I'll see if I can improve that & then share.

josesimoes, 2022-10-25:
If you are already working on a fix, even better. Nothing against improving our json library.
I guess you're already using the benchmark tests, therefore your comment on the performance.

If you can't find a nice way to tackle it, it's probably best to have a new config property that a developer can enable for the library to parse unicode. Default would be as it is for performance reasons. Happy to discuss further and review your code.

closed 2022-10-27 by josesimoes