nanoframework/Home #1173: defect in System.Device.DelayHelper

yureng, 2022-10-27:
### Library/API/IoT binding

System.Diagnostics.Stopwatch

### Visual Studio version

VS2022 (17.3.6)

### .NET nanoFramework extension version

2022.2.0.44

### Target name(s)

ESP32_BLE_REV0

### Firmware version

1.8.0.534

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4.2
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Sep 28 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM, support for BLE
  CLR Version: 1.8.0.534
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Kinghonor.SmartAir, 1.0.0.0
  nanoFramework.Runtime.Events, 1.10.0.3
  nanoFramework.Hardware.Esp32, 1.4.1.63006
  System.IO.Streams, 1.1.15.36091
  nanoFramework.System.Text, 1.2.7.60109
  nF.Modbus, 1.0.0.0
  nanoFramework.System.Collections, 1.4.0.3
  System.Math, 1.4.4.3
  System.IO.FileSystem, 1.1.9.8185
  mscorlib, 1.12.0.4
  System.Device.Gpio, 1.1.18.20780
  System.Diagnostics.Stopwatch, 1.2.0.0
  System.IO.Ports, 1.1.25.11551
  nanoFramework.Runtime.Native, 1.5.4.3
  Iot.Device.Tm1637, 1.2.0.0
  nanoFramework.Runtime.Events, 1.11.1.42088
  Kinghonor.SmartAir.Standard, 1.0.0.0
  Kinghonor.SmartAir.Common, 1.0.0.0

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.Bluetooth v100.0.2.0, checksum 0x066AC1EC
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x00013800
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

When I used the IoT.Device.Tm1637 library to drive the TM1637 digital Segment, I found that the component response was very slow, which was not a normal state at all

The reason is that the [TM1637](https://github.com/nanoframework/nanoFramework.IoT.Device/blob/develop/devices/Tm1637/Tm1637.cs)  call error of the function System.Device.DelayHelper.DelayMicroseconds(ClockWidthMicroseconds, true) in the TM1637.cs file takes a lot of time

In the [System.Device.DelayHelper.cs](https://github.com/nanoframework/nanoFramework.IoT.Device/blob/2c5e0e484daa1f5591129220b7bb759254108930/src/devices_generated/Common/System/Device/DelayHelper.cs#L68) file, there is the following code (lines 68 - 92):

```
public static void DelayMicroseconds(int microseconds, bool allowThreadYield)
{
            var time = TimeSpan.FromTicks(microseconds * TicksPerMicrosecond);
            Delay(time, allowThreadYield);
 }

public static void DelayMilliseconds(int milliseconds, bool allowThreadYield)
{
            /* We have this as a separate method for now to make calling code clearer
             * and to allow us to add additional logic to the millisecond wait in the
             * future. If waiting only 1 millisecond we still have ample room for more
             * complicated logic. For 1 microsecond that isn't the case.
             */

            var time = TimeSpan.FromTicks(milliseconds * TicksPerMillisecond);
            Delay(time, allowThreadYield);
 }
```

After the code is decompiled, it looks like this：

```
 public static void DelayMicroseconds(int microseconds, bool allowThreadYield)
  {
            TimeSpan time = TimeSpan.FromTicks((long)microseconds * 10L);
            Delay(time, allowThreadYield);
 }

public static void DelayMilliseconds(int milliseconds, bool allowThreadYield)
 {
            TimeSpan time = TimeSpan.FromTicks((long)milliseconds * 10000L);
            Delay(time, allowThreadYield);
 }
```

It looks like this will cause DelayMilliseconds(1) to delay longer than DelayMicroseconds(1)  ！！！
I'm not sure if my understanding is correct.

### Screenshots

[image]

Ellerbach, 2022-10-27:
1 millisecond = 1000 microseconds
So looking at what you point out, it seems right like this. Something to keep in mind as well is that you will not get a microsecond precision in .NET nanoFramework.
Can you maybe measure both function with a simple time difference? 
for the microsecond one make it long like 20_000 microseconds and like 20 milliseconds?
Thanks!

yureng, 2022-10-27:
> 

Sorry, it was I who was confused for a moment and misunderstood. Thank you very much for your reply !

closed 2022-10-27 by yureng

Ellerbach, 2022-10-28:
> Sorry, it was I who was confused for a moment and misunderstood. Thank you very much for your reply !

No worry :-) We are here to help ;-) Bugs happened!