nanoframework/Home #1174: Critical bug in json library

josesimoes, 2022-10-27:
### Library/API/IoT binding

json library

### Description

Sonarcloud is reporting a bug that needs to be addressed ASAP:
https://sonarcloud.io/project/issues?resolved=false&sinceLeakPeriod=true&types=BUG&id=nanoframework_lib-nanoFramework.Json&open=AYQKFaUbLqvlGbRq34ta

Ellerbach, 2022-10-27:
It's in the test and if null, then the test will fail as it's not supposed to be null :-) So not sure there is anything to fix here.

josesimoes, 2022-10-27:
I've overlooked that it was on a test. 😊 
Despite the assert before that line will fail, it shouldn't continue the execution if it's null otherwise it will throw an exception, right?

Ellerbach, 2022-10-27:
Absolutely and the test will fail :-) Meaning something is wrong. I though I setup SonarCloud not to check the test directory but in all then new repo I created and all those I've been touching by adding tests but that may not have been the case for this one!
So just add the folder where the tests are in the ignore settings in the SonarCloud project and should be all fine ;-)

josesimoes, 2022-11-18:
Fixed as part of nanoframework/nanoFramework.Json#280

closed 2022-11-18 by josesimoes