nanoframework/Home #1175: Rounding to 2 decimal point not working

alberk8, 2022-10-30:
### Library/API/IoT binding

nanoframework.System.Math

### Visual Studio version

VS2022 17.3.6

### .NET nanoFramework extension version

2022.2.0.44

### Target name(s)

ESP32_REV3

### Firmware version

1.8.0.699

### Description

Rounding a double to two(2) decimal places does not work as expected.  The results is the same as the source. 

### How to reproduce

```
double tempValue = 36.3999999d;
double roundedValue = Math.Round(tempValue * 100d) / 100d;
Console.WriteLine("Rounded Value: " + roundedValue);
Console.WriteLine("Rounded Value to String : " + roundedValue.ToString());
```

### Expected behaviour

There result should be 36.4.   This result is confirmed correct on .Net 6.

Ellerbach, 2022-10-30:
This is normal as depending on the platform you have a float behind the scene of the double.
Try it with 36.4d of 36.4000001d and quite some equivalents, you'll always get back 36.3999999.
float like double are not precise numbers. There are numbers you'll never be able to get exact regardless of nano or full .NET. I don't have some of those in minds but you could find much much much examples like this.
Proper way of rounding to any decimal with a minimum effect::

```csharp
double tempValue = 36.3999999d;
// This to avoud rounding up, it will always round to the entier part
double roundedValue = Math.Round(tempValue - 0.5);
double roundedValueAfter = Math.Abs(Math.Round((roundedValue - tempValue) * 100d));
Console.WriteLine($"Rounded Value to String: {roundedValue}.{roundedValueAfter}");
```

Again, not related to nano, just related to internal usage of float in some platforms and this phenomenon appear for a lot of numbers.

TerryFogg, 2022-10-30:
The problem here is the .ToString() operation on the double.
Single stepping this code sees that the answer is correct but the output is not

closed 2022-10-31 by TerryFogg

alberk8, 2022-10-31:
I have downgraded to Float.  it seems to work correctly with rounding.

TerryFogg, 2022-10-31:
> 

Down grading may work for this but does it work for all cases?
Without context, I don't know what you are trying to achieve, another Round() of the result to the number of places you require?
Obviously, you need to take into account experimental errors and significant digits in measurements

alberk8, 2022-10-31:
> > 
> 
> Down grading may work for this but does it work for all cases? Without context, I don't know what you are trying to achieve, another Round() of the result to the number of places you require? Obviously, you need to take into account experimental errors and significant digits in measurements

What I am trying to do is to round a float or double to two decimal place.  I don't have anything beside the ESP32,  if you can test on other Architecture that would be nice.  Thank you.

```
   float f1 = 34.75545f;
   float f2 = (float) Math.Round(f1 * 100) ;
   float f3 = (float)Math.Round(f1 * 100)/100;

   Console.WriteLine($"Float F1: {f1}");
   Console.WriteLine($"Float F2: {f2}");
   Console.WriteLine($"Float F3: {f3}");
```

TerryFogg, 2022-10-31:
Some of these work some don't 

            double number = 7861.123456;
            Console.WriteLine($"Number: {number}"); // Number: 7861
            Console.WriteLine($"C -> {number:C}");  // $7,861.00
                                                    // Precision specifier indicates the
                                                    // desired number of decimal places.
            Console.WriteLine($"C0 -> {number:C0}"); // $7,861
            Console.WriteLine($"C1 -> {number:C1}"); // $7,861.0
            Console.WriteLine($"C2 -> {number:C2}"); // $7,861.00
            Console.WriteLine($"C3 -> {number:C3}"); // $7,861.000

            Console.WriteLine($"D -> {number:D}");   // 7861
            Console.WriteLine($"D3 -> {number:D3}"); // 7861
            Console.WriteLine($"D4 -> {number:D4}"); // 7861
            Console.WriteLine($"D5 -> {number:D5}"); // 07861
            Console.WriteLine($"D6 -> {number:D6}"); // 007861

            Console.WriteLine($"E -> {number:E}"); // 7.861000E+003
            Console.WriteLine($"e -> {number:e}"); // 7.861000E+003
            Console.WriteLine($"F -> {number:F}"); // 7861.00
                                                   // Precision specifier indicates the
                                                   // desired number of decimal places.
            Console.WriteLine($"F0 -> {number:F0}"); // 7861
            Console.WriteLine($"F1 -> {number:F1}"); // 7861.0
            Console.WriteLine($"F2 -> {number:F2}"); // 7861.00
            Console.WriteLine($"N -> {number:N}");   // 7,861.00
                                                     // Precision specifier indicates the
                                                     // desired number of decimal places.
            Console.WriteLine($"N0 -> {number:N0}"); // 7,861
            Console.WriteLine($"N1 -> {number:N1}"); // 7,861.0
            Console.WriteLine($"N2 -> {number:N2}"); // 7,861.00
            Console.WriteLine($"P -> {1:P}");        // 100.00%
            Console.WriteLine($"P0 -> {1:P0}");      // 100%
            Console.WriteLine($"X -> {10995:X}");    // 2AF3

            // Formatted with single decimal place.
            Console.WriteLine($"2 -> {786.1:000.0}"); // 786.1
                                                      // Display zero, if the number includes fewer
                                                      // decimal places then the custom specifier.
            Console.WriteLine(
                $"3 -> {786.1:000.00}"); // 786.10
                                         // Number is rounded if includes more decimal
                                         // places then the custom specifier.
            Console.WriteLine($"4 -> {786.128:000.00}"); // 786.13

            // Digit Placeholder(#)

            Console.WriteLine($"1 -> {78.7:0.#}"); // 78.7
            Console.WriteLine($"2 -> {1.7:#0.0}"); // 1.7
            Console.WriteLine($"3 -> {2.7:0#.#}"); // 02.7
            Console.WriteLine($"4 -> {2.7:0##.#}"); // 002.7
                                                    // Number is rounded if includes more decimal places then the custom specifier.
            Console.WriteLine($"5 -> {786.36:0.#}"); // 786.4
            Console.WriteLine($"6 -> {786.348:0.##}"); // 786.35
            Console.WriteLine($"7 -> {786.8:0##}"); // 787

            //Thousands Separator( , )

            Console.WriteLine($"1 -> {78612.3:#,#.#}"); // 78,612.3
            Console.WriteLine($"2 -> {78612.3:#,#}"); // 78,612

            // Percentage Placeholder( % )

            // Multiplied by 100 before it is formatted
            Console.WriteLine($"1 -> {.7861:#00.##%}"); // 78.61%

            // Section Separator ( ; )
            Console.WriteLine($"1 -> {7.71:+0.00;-0.00;0}"); // +7.71
            Console.WriteLine($"2 -> {-8.81:+0.00;-0.00;0}"); // -8.81
            Console.WriteLine($"3 -> {0.0:+0.00;-0.00;0}"); // 0

            Console.WriteLine($"4 -> {7:+#;-#;0}"); // +7
            Console.WriteLine($"5 -> {-8:+#;-#;0}"); // -8
            Console.WriteLine($"6 -> {0:+#;-#;0}"); // 0

Number: 7861.123456
C -> C
C0 -> C0
C1 -> C1
C2 -> C2
C3 -> C3
D -> D
D3 -> D3
D4 -> D4
D5 -> D5
D6 -> D6
E -> E
e -> e
F -> 7861.12
F0 -> 7861
F1 -> 7861.1
F2 -> 7861.12
N -> 7,861.12
N0 -> 7,861
N1 -> 7,861.1
N2 -> 7,861.12
P -> P
P0 -> P0
X -> 2AF3
2 -> 000.0
3 -> 000.00
4 -> 000.00
1 -> 0.#
2 -> #0.0
3 -> 0#.#
4 -> 0##.#
5 -> 0.#
6 -> 0.##
7 -> 0##
1 -> #,#.#
2 -> #,#
1 -> #00.##%
1 -> +0.00;-0.00;0
2 -> +0.00;-0.00;0
3 -> +0.00;-0.00;0
4 -> +#;-#;0
5 -> +#;-#;0
6 -> +#;-#;0

TerryFogg, 2022-10-31:
This works, seems to be plenty of examples out there, but not all would work.
This one does
            double d1 = 34.75545667d;
            double answer2 = double.Parse(d1.ToString("f2"));  --> 34.75
            double answer3 = double.Parse(d1.ToString("f3"));  --> 34.755
            double answer4 = double.Parse(d1.ToString("f4"));  --> 34.7555