nanoframework/Home #1176: Add method to DeviceClient for sending response to direct methods

pheeblyboos, 2022-11-01:
### Description

We're currently working on a solution using NanoFramework where we need to restart our devices during a direct method request, which prevents us to immediately report back to the direct method, causing it to report failure on the IoT Hub end. The documentation from Microsoft states it is possible to respond back to a specified direct method through MQTT later by communicating to the correct endpoint with the provided rid (https://learn.microsoft.com/en-us/azure/iot-hub/iot-hub-devguide-direct-methods#response-1).
 

### How to solve the problem

It would be great if it is possible to either use the current connection that is being leveraged by the deviceclient or if there is an method available to publish information to a direct method endpoint. As far as I've studied your code so far, that currently doesn't seem possible.
Is there a chance that this will be added in the future?

### Describe alternatives you've considered

We tried to instead update the device twin to read return statuses, but this is not ideal since we found the twin updates not to be 100% reliable.

josesimoes, 2022-11-01:
Why don't you fire a Thread to reboot the device with a small Sleep at the very beginning to allow enough time for mqtt to report back before actually rebooting the device? 🤔

pheeblyboos, 2022-11-01:
Great suggestion, that also works :') 
Thanks for thinking along :)

closed 2022-11-01 by pheeblyboos