nanoframework/Home #1181: .NET nanoFramework Extension for VS Code v1.0.130^ missing nanoff.dll

Sour-Codes, 2022-11-07:
### Tool

Visual Studio Code extension

### Description

v1.0.30 and higher missing nanoff.dll yet `nanoFramework: Flash Device` from command palette attemts to use said file.

### How to reproduce

1. Install v1.0.30 or higher of the extension
2. Attempt `nanoFramework: Flash Device`
3. Be disappointed by "Possible reasons" including "\nanoframework.vscode-nanoframework-1.0.132/dist/utils/nanoFirmwareFlasher/nanoff.dll does not exist."
4. Install v1.0.127 or older and see you aren't going insane and the file is just missing from v1.0.130 and newer

### Expected behaviour

Never seeing the following:

> Could not execute because the specified command or file was not found.
> Possible reasons for this include:
>   * You misspelled a built-in dotnet command.
>   * You intended to execute a .NET program, but dotnet-c:\Users\user\\.vscode\extensions\nanoframework.vscode-nanoframework-1.0.132/dist/utils/nanoFirmwareFlasher/nanoff.dll does not exist.
>   * You intended to run a global tool, but a dotnet-prefixed executable with this name could not be found on the PATH.

### Screenshots

[image]

Sour-Codes, 2022-11-07:
I tried a cheap workaround by just copying nanoff.dll to the folder, but it's now complaining that hostpolicy.dll is missing too. 
 I'll just stick to an older version that's working for now rather than playing the missing file lottery as don't want to expend any luck from the $1.9B lotto chance tonight.

OaimIsGnay, 2022-11-11:
I also have this problem, and the deployment will not find the device

networkfusion, 2022-11-13:
Please check if any of the recent changes have fixed this an report back: https://github.com/nanoframework/nf-VSCodeExtension/commits/main

networkfusion, 2022-11-13:
@josesimoes See comment above... Is it possible we can release a new extension based on that?

josesimoes, 2022-11-14:
@networkfusion maybe it's because is Monday morning for me but... what are those 130 and 132 that you're referring to? 😅

Sour-Codes, 2022-11-17:
@josesimoes 
[image]

OaimIsGnay, 2022-11-22:
> @josesimoes [image]

Hello, which version can solve this problem?

Sour-Codes, 2022-11-22:
@OaimIsGnay I'm using 1.0.98 until this is resolved.

OaimIsGnay, 2022-11-23:
> 

[image]
Device not found.
I don't know why

hinnerkweiler, 2023-11-09:
Since the first attempt to solve this was blown off, is there any progress? 👯

Ellerbach, 2023-11-16:
Found the root cause of the flash failing with the latest VS Code version. It's related to changes made in nanoff build. Now, the challenge is to have the time to fix those and do couple of tests to see which options will work best:

* ask the user to install nanoff and add it properly to the path
* have a specific version of nanoff install in the extension but from the dotnet tool
* have a specific built of nanoff coming with the extension like today

The last point is what we've been using, it has quite a lot of drawbacks. So I4ll investigate the 2 first ones. We anyway ask the user to install mono, so why not nanoff as well.
That would definitely simplify a lot of things.

adrianoc, 2023-12-20:
On my box (Ubuntu 22.04, with .Net 7 & 8 sdks installed) version 1.0.115 is the newest one that successfully builds and deploy to my ESP32 device.

btw, I've tried to change the extension to use `nanoff` from the `.net global install folder` but `nanoff` returns an error: 

`Error E2002: Error executing operation with nano device. (Couldn't find the deployment file at the specified path.)`

The command line used  by the extension: `/home/adriano/.dotnet/tools/nanoff --nanodevice --deploy --serialport  /dev/ttyUSB0 --image /home/adriano/Projects/study/DotNet/NanoframeworkTest/OutputDir/ -v d`

junalmeida, 2024-04-09:
Is the vscode extension abandoned? 

It seems this error is more than 1yr old.

networkfusion, 2024-04-09:
> Is the vscode extension abandoned?
> 
> It seems this error is more than 1yr old.

Not abbandoned, but needs some TLC by a willing developer.

junalmeida, 2024-04-10:
Is there any viable workaround? Calling `nanoff` manually after a successful build with the same parameters doesn't work either, failing on a missing deployment file

josesimoes, 2024-04-10:
nanoff is working as expected... no issues reported about it.
I've been using it myself without any issues.
Are you on our Discord server? It's so much easy to provide support there, instead of piling comments on github issues.

networkfusion, 2024-04-12:
> nanoff is working as expected... no issues reported about it. I've been using it myself without any issues. Are you on our Discord server? It's so much easy to provide support there, instead of piling comments on github issues.

Pretty sure this is on linux / MAC.

junalmeida, 2024-05-02:
@networkfusion @josesimoes yep, it doesn't work on MacOS Sonoma (intel)

closed 2024-06-19 by josesimoes