nanoframework/Home #1184: DateTime.ToString(string format) returns wrong time

DejanBracun, 2022-11-13:
### Library/API/IoT binding

nanoFramework.CoreLibrary.1.12.0

### Description

DateTime.ToString(string format) does not work correctly.

### Expected behaviour

I would expect that DateTime.ToString() and DateTime.ToString(string format) would return the same result (code attached bellow).
The program prints current time every 5 seconds. At each output DateTime.ToString() is increesing time for 5 seconds (as expected) while DateTime.ToString(string format) for only 0.01 second (not expected). 

### Sample project or code

public static void Main()
{
   while (true)
   {
        var time = DateTime.UtcNow;
        Debug.WriteLine("==================================");
        Debug.WriteLine("ToString()                " + time.ToString());
        Debug.WriteLine("ToString(string format)   " + time.ToString("MM/dd/yyyy HH:mm.ff"));
        Thread.Sleep(5000);
    }
}

// Console output:
// ==================================
// ToString()                11/13/2022 17:03:22
// ToString(string format)   11/13/2022 17:03.61
// ==================================
// ToString()                11/13/2022 17:03:27
// ToString(string format)   11/13/2022 17:03.62
// ==================================
// ToString()                11/13/2022 17:03:32
// ToString(string format)   11/13/2022 17:03.63
// ==================================
// ToString()                11/13/2022 17:03:37
// ToString(string format)   11/13/2022 17:03.64

josesimoes, 2022-11-13:
If you're testing for the same output you should be using the format string `"MM/dd/yyyy HH:mm.ff" `.
Note the use of `ss` instead of ff for the seconds.

networkfusion, 2022-11-13:
Test case suggests that you are not using the correct input:

```            while (true)
            {
                var time = DateTime.UtcNow;
                Debug.WriteLine("==================================");
                Debug.WriteLine($"ToString()                        -> {time.ToString()}");
                Debug.WriteLine($"ToString(\"o\")                   -> {time.ToString("o")}");
                Debug.WriteLine($"ToString(\"u\")                   -> {time.ToString("u")}");
                Debug.WriteLine($"ToString(\"MM/dd/yyyy HH:mm.ff\") -> {time.ToString("MM/dd/yyyy HH:mm.ff")}");
                Debug.WriteLine($"ToString(\"MM/dd/yyyy HH:mm.ff\") -> {time.ToString("MM/dd/yyyy HH:mm.ss")}");
                Thread.Sleep(5000);
            }
```

Output on an STMF769I:

```
==================================
ToString()                        -> 01/01/1980 00:10:57
ToString("o")                   -> 1980-01-01T00:10:57.2580000Z
ToString("u")                   -> 1980-01-01 00:10:57Z
ToString("MM/dd/yyyy HH:mm.ff") -> 01/01/1980 00:10.25
ToString("MM/dd/yyyy HH:mm.ff") -> 01/01/1980 00:10.57
==================================
```

closed 2022-11-13 by networkfusion