nanoframework/Home #1187: Failure to flash nanoframework onto ESP32 Lilygo

FraserElectronics, 2022-11-18:
### Tool

nanoff

### Description

Trying to update ESP32 Lilygo module with nanoframework version 1.8.0.581 using nanoff version nanoff 2.4.2+c4df2f6716. 
Flashing device is successful but looking at ESP32 on TeraTerm shows a continual reboot cycle with the following debug output.

```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46
```

### How to reproduce

1. Download firmware version 1.8.0.581 from cloudsmith and unzip into a directory.
2. Flash device ```nanoff --update --target ESP32_LILYGO --serialport COMx --fwversion 1.8.0.581```
3. Connect to device using serial terminal such as TeraTerm @ 115200 baud.
4. Observe boot messages.

### Expected behaviour

Expect device to be ready to accept .NET application.

josesimoes, 2022-11-18:
@FraserElectronics I'm testing this with a ESP32_PICO image (which is the closest I have) and I can't reproduce this...
Can you please run nanoff again this time adding `-v  diag` and copy & paste here the complete output of the operation?

Also please add here the complete output of the boot log from IDF.

FraserElectronics, 2022-11-18:
Here is the output from nanoff

```
C:\temp\nanoframework\esp32\lilygo\twoweeksago>nanoff --update --target ESP32_LILYGO --serialport COM20 --fwversion 1.8.0.581 -v diag
.NET nanoFramework Firmware Flasher v2.4.2+c4df2f6716
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM20 @ 1500000 baud to connect to ESP32.
Reading details from chip...

Executing esptool with the following parameters:
'--port COM20  --chip auto   --after no_reset_stub flash_id'

OK

Connected to:
ESP32 (ESP32-PICO-V3-02 (revision 3))
Features WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Flash size 8MB unknown from  (manufacturer 0x104 device 0x16407)
PSRAM: not available
Crystal 40MHz
MAC AC:0B:FB:6C:D0:C8

Trying to find ESP32_LILYGO in stable repository...OK
Extracting ESP32_LILYGO-1.8.0.581.zip...OK

Updating to 1.8.0.581

Erasing flash...

Executing esptool with the following parameters:
'--port COM20  --chip esp32   --after no_reset_stub erase_flash'

OK
Flashing firmware...

Executing esptool with the following parameters:
'--port COM20 --baud 1500000 --chip esp32   --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x1000 "C:\Users\mail\.nanoFramework\fw_cache\ESP32_LILYGO\bootloader.bin" 0x10000 "C:\Users\mail\.nanoFramework\fw_cache\ESP32_LILYGO\nanoCLR.bin" 0x8000 "C:\Users\mail\.nanoFramework\fw_cache\ESP32_LILYGO\partitions_8mb.bin"'

Wrote 23888 bytes (15082 compressed) at 0x00001000 in 0.2 seconds (effective 919.0 kbit/s)...                 e None
Hash of data verified.
Wrote 1119824 bytes (723640 compressed) at 0x00010000 in 8.7 seconds (effective 1026.7 kbit/s)...
Hash of data verified.
Wrote 3072 bytes (136 compressed) at 0x00008000 in 0.0 seconds (effective 778.3 kbit/s)...
Hash of data verified.
OK
```

When you say boot log from IDF, do you mean the serial terminal or do I need to use the IDF tool?

Andy

alberk8, 2022-11-19:
Basically set the baud rate to 115200 on TeraTerm and press the reset button on the Device.  There should be a boot message/log.

FraserElectronics, 2022-11-19:
Here is the serial output as posted on Discord.

```
st:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46
```

Andy

FraserElectronics, 2022-11-19:
Sorry, forgot to include the initial boot messages which are below:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5684
load:0x40078000,len:14432
ho 0 tail 12 room 4
load:0x40080400,len:3684
entry 0x40080678
I (30) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v4.4.2
I (30) boot: build Oct 31 2022 15:06:35
I (31) boot: chip revision: 3
I (35) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 8MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 001a0000
I (94) boot:  3 deploy           Unknown data     01 84 001b0000 002e0000
I (102) boot:  4 config           Unknown data     01 82 00490000 00200000
I (109) boot: End of partition table
I (113) boot_comm: chip revision: 3, min. application chip revision: 0
I (121) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a394h (172948) map
I (191) esp_image: segment 1: paddr=0003a3bc vaddr=3ffb0000 size=0353ch ( 13628) load
I (197) esp_image: segment 2: paddr=0003d900 vaddr=40080000 size=02718h ( 10008) load
I (201) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=c8650h (820816) map
I (500) esp_image: segment 4: paddr=00108678 vaddr=40082718 size=18f24h (102180) load
I (542) esp_image: segment 5: paddr=001215a4 vaddr=400c0000 size=00068h (   104) load
I (542) esp_image: segment 6: paddr=00121614 vaddr=50000000 size=00010h (    16) load
I (560) boot: Loaded app from partition at offset 0x10000
I (561) boot: Disabling RNG early entropy source...
I (573) psram: This chip is ESP32-PICO-V3-02
I (573) spiram: Found 16MBit SPI RAM device
I (573) spiram: SPI RAM mode: flash 80m sram 40m
I (578) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (586) cpu_start: Pro cpu up.
I (589) cpu_start: Starting app cpu, entry point is 0x400859c8
I (0) cpu_start: App cpu up.
I (611) cpu_start: Pro cpu start user code
I (611) cpu_start: cpu freq: 240000000
I (611) cpu_start: Application information:
I (616) cpu_start: Project name:     nanoCLR
I (621) cpu_start: App version:      1.8.0.581
I (626) cpu_start: Compile time:     Oct 31 2022 15:06:09
I (632) cpu_start: ELF file SHA256:  f9b2cdd93607d359...
I (638) cpu_start: ESP-IDF:          v4.4.2
I (643) heap_init: Initializing. RAM available for dynamic allocation:
I (650) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (656) heap_init: At 3FFBC310 len 00023CF0 (143 KiB): DRAM
I (662) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (669) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (675) heap_init: At 4009B63C len 000049C4 (18 KiB): IRAM
I (681) spiram: Adding pool of 2047K of external SPI memory to heap allocator
I (690) spi_flash: detected chip: boya
I (694) spi_flash: flash io: dio
I (709) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (710) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid heáets Jul 29 2019 12:21:46
```

Andy

josesimoes, 2022-11-21:
@FraserElectronics  can you please give a try with the target name "LilygoTWatch2021" ? 
This build is for rev3 pico devices and so it should be better suited.

josesimoes, 2022-11-22:
@FraserElectronics yes, that's the expected output. It's booting into the CLR now!
It's probably wise to add another sanity check to nanoff to prompt users that the fw version is not suitable in cases like this...

FraserElectronics, 2022-11-22:
@josesimoes Just built and ran a simple 'Blinky' program and it downloaded and ran fine 😊
I will leave you to close this issue if you feel it is appropriate to do so.

Andy

josesimoes, 2022-11-22:
Awesome! Thanks for reporting back. 👍🏻 
Let's leave it open to track that improvement on nanoff.

josesimoes, 2022-11-28:
Because there is a specific target being buit for Lilygo, which requires calling nanoff with the proper target name, it's hard to code this check. 

Closing as this sanity check it's not trivial to implement and it's better relying on the user to use the correct target name when using nanoff.

closed 2022-11-28 by josesimoes