nanoframework/Home #1188: Crash when selecting Application tab on project settings

FraserElectronics, 2022-12-06:
### Tool

Visual Studio extension

### Description

When opening Application tab on project settings it crashes with the error:

The source completed without providing data to receive.

This is not a show stopper for me - just thought I would mention it.

Andy

### How to reproduce

1. Select Project -> Properties.
2. Click on Application tab.

### Expected behaviour

Not to crash?

### Screenshots

[image]

### Aditional context

Visual Studio 2022 Version 17.4.2
Latest Nano framework extension.

josesimoes, 2022-12-06:
Most likely a duplicate of #1022.
We're using a modified project system from the regular C# one and this is bitting us every now and then...

What are you trying to acomplish in that tab? If you need to tweak anything on the project, you're better editing nfproj file.

FraserElectronics, 2022-12-06:
@josesimoes Jose, I was merely looking through the properties and stumbled across it. Sorry, I should have looked for similar reported issues as you are right, it is a duplicate of #1022. I will close out this issue.

closed 2022-12-06 by FraserElectronics