nanoframework/Home #1194: ESP32_PICO SerialPort throws an NativeInit error on Open()

Feiko, 2022-12-15:
### Library/API/IoT binding

System.IO.Ports

### Target name(s)

ESP32_PICO

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 6407ecb
  Target:   ESP32_PICO
  Platform: ESP32

Firmware build Info:
  Date:        Nov 16 2022
  Type:        MinSizeRel build, support for PSRAM
  CLR Version: 1.8.0.750
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  nfTestSerial, 1.0.0.0
  nanoFramework.System.Text, 1.2.22.3995
  mscorlib, 1.12.0.4
  nanoFramework.Runtime.Events, 1.11.1.42088
  nanoFramework.Hardware.Esp32, 1.4.8.26232
  System.IO.Streams, 1.1.27.27650
  System.IO.Ports, 1.1.46.59502

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffc66bc  0x00018800
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

When you try to create and open a serial port on a ESP32_PICO (I used an m5 Atom Lite), you get an exeption when calling SerialPort.Open(). The code runs fine on a ESP_WROOM_32.

I get the following exception:
    ++++ Exception System.Exception - CLR_E_PIN_UNAVAILABLE (1) ++++
    ++++ Message: 
    ++++ System.IO.Ports.SerialPort::NativeInit [IP: 0000] ++++
    ++++ System.IO.Ports.SerialPort::Open [IP: 000f] ++++
    
 I also tried running the code with the following specific pins:
```
           Configuration.SetPinFunction(26, DeviceFunction.COM2_TX);
           Configuration.SetPinFunction(32, DeviceFunction.COM2_RX);
```

In the past I had no probles executing this code on a ESP32_PICO. I have tried multiple devices. 

 

### How to reproduce

Use the sample code below and run on a ESP32_PICO 

### Sample project or code

```
private static SerialPort _serialDevice;

public static void Main()
        {
            Debug.WriteLine("Hello from nanoFramework!");

            _serialDevice = new SerialPort("com2", 115200, Parity.None, 8, StopBits.One);
            _serialDevice.Open();

            Debug.WriteLine("serial open!");

            Thread.Sleep(Timeout.Infinite);
        }
```

Ellerbach, 2022-12-15:
@Feiko that seems very logical :-)

In your code, there is no pin setup for COM2. There is no default one, so it will throw an exception.
See: https://docs.nanoframework.net/content/esp32/esp32_pin_out.html

Setup pins and you'll be good to go.

romankiss, 2022-12-23:
@Feiko, it looks like is a bug after firmware version **1.8.0.699**, try to go back to the _--fwversion 1.8.0.699_

Thanks
Roman

Feiko, 2022-12-29:
@romankiss Thanks for the suggestion. I can indeed confirm that my code works fine on firmware version 1.8.0.699.

@Ellerbach  Thanks for suggesting the settings for the pins for com2. As mentioned earlier, I already used the pin settings. 
I only have this issue when using the PICO the WROOM works fine. 
Here my exact code:

```
using nanoFramework.Hardware.Esp32;
using System.Diagnostics;
using System.IO.Ports;
using System.Threading;

namespace nfTestSerial
{
    public class Program
    {
        private static SerialPort _serialDevice;
        public static void Main()
        {
            Configuration.SetPinFunction(26, DeviceFunction.COM2_TX);
            Configuration.SetPinFunction(32, DeviceFunction.COM2_RX);

            _serialDevice = new SerialPort("COM2", 115200, Parity.None, 8, StopBits.One);
            _serialDevice.Open();
            Debug.WriteLine("serial open!");
            Thread.Sleep(Timeout.Infinite);
        }
    }
}
```

josesimoes, 2022-12-29:
The only major change I can see after that release was adding support for I2S, which is not even enabled for the PICO...

Ellerbach, 2022-12-29:
So trying to follow the path of execution, it goes to the native init: https://github.com/nanoframework/nf-interpreter/blob/c4fafe5b06e669c48d7261df703fe3195239fa6b/targets/ESP32/_nanoCLR/System.IO.Ports/sys_io_ser_native_System_IO_Ports_SerialPort.cpp#L838

This function is called as well: https://github.com/nanoframework/nf-interpreter/blob/c4fafe5b06e669c48d7261df703fe3195239fa6b/targets/ESP32/_nanoCLR/System.IO.Ports/sys_io_ser_native_System_IO_Ports_SerialPort.cpp#L948

There are few places in this code where you can get an exception. But it should not happen in the `NativeConfig___VOID` otherwise, you'll see it in the traces. So it's leaving 4 other places in the NativeInit one.

Good would be to place a tracer to output in the debug and check the progression of the code to identify where it breaks. That require to rebuild the image and flash it.
Best is of course to debug this and check as well.

josesimoes, 2023-01-02:
I've updated my PICO to ESP32_PICO-1.8.0.858
The exception I'm getting is:
```
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
    ++++ Exception System.Exception - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.IO.Ports.SerialPort::NativeInit [IP: 0000] ++++
    ++++ System.IO.Ports.SerialPort::Open [IP: 000f] ++++
Exception thrown: 'System.Exception' in System.IO.Ports.dll
```

Feiko, 2023-01-09:
Reopening the issue because there still seems to be a memory issue. I need to have a larger Readbuffer to store a complete serial message that is send every second. The setting:
`_serialDevice.ReadBufferSize = 3072; `

Throws the same exception as reported before. Again I only get this exception for the PICO.

romankiss, 2023-01-09:
@Feiko, you should use the port receiving small buffer for balancing between the base network stream and application large input buffer. The default size of the port _ReadBuffer_ is 256 bytes and it's working well with 512 bytes also.

Using the eventing handler (__serial.DataReceived_)  and either the _serial.ReadExisting() or _serial.ReadByte() you can received (and have a control over receiving process) a large serial message instead of the _serial.ReadLine(), where the port receiving and storing to the ReadBuffer all bytes and waiting for the _serial.NewLine value.

However, reopening issue for SerialPort is still valid, see my yesterday comment [here](https://github.com/nanoframework/nf-interpreter/pull/2509).

Thanks
Roman

Feiko, 2023-01-10:
@romankiss I wrote this implementation for reading the output from a Smart Meter that are installed in the main socket box of most Dutch homes. I've tried using events to read smaller chunks. The problem was that the message, that is send each second, is quite large and the mechanisms to use events was to slow. Therefor events started stacking up and data was lost because the buffer ran out. Eventually after 2 or 3 messages the device ran out of memory completely (probably because of the event backlog). For this reason I asked for the feature to create a bigger buffer. I have a timer that checks the message every second. Not the most elegant solution, I admit. But with a lot of trial and error the only solution that worked for my scenario. 
Also this worked back in the day ;-)

romankiss, 2023-01-11:
@Feiko  What's your baud rate? For instance, when the baudrate = 115200, the serial port received 115 bytes each 10 ms and in the case of _serial.ReadBuffer=512 bytes, the event handler has enough time (more than 40 ms) to move bytes to the large application input buffer. Of course, the size of application input buffer is limited by device resource capability. 

Thanks
Roman

Feiko, 2023-01-12:
@romankiss, Thanks for thinking along. The baudrate is indeed 115200. That would indeed give me 40ms to read the buffer. As I remember the _serial.DataReceived was raised continuously and stacking up while receiving data. This made the system slow and run out of memory. Like I said I've tried to use this event, not very keen to revisit this. However I must admit that I used this with 256 buffer not the 512 bytes. If I remember correctly every thread is allocated 10ms time slot. So if during the 40ms multiple DataReceived events are raised (each on a separate thread), this can start taking up a lot of resources quite quick. I also need to handle the received data and send it to a server. A year ago I had this working with no resources to spare by using the enlarged buffer.

josesimoes, 2023-02-23:
@Feiko just tested this with the SerialCommunication on a PICO MATRIX and everything went fine...
Can you please give it another try to check if any of the recent changes had any effect on this?

romankiss, 2023-02-24:
@josesimoes it looks like the issue of opening two SerialCommunication ports on a PICO MATRIX/LITE is still there.

Thanks
Roman

romankiss, 2023-02-27:
@josesimoes, Thank you so much. I have just successfully tested two SerialCommunication ports with a ReadBufferSize=3500,
ESP32_PICO version 1.8.1.13
 nanoFramework.AtomLite version 1.1.119

Thanks again.
Roman

josesimoes, 2023-02-28:
@romankiss glad to know that. Most likely because of the increase in reserved memory for IDF allocation added today.

@Feiko care to retest this at your end?

josesimoes, 2023-09-01:
@Feiko lacking feedback from you, I'm closing this as it should be fixed since February. 
If that's not the case for you feel free to reopen the issue and/or report your findings here.

closed 2023-09-01 by josesimoes