nanoframework/Home #1195: Every json string crashes the logger

DaveSchmid, 2022-12-18:
### Library/API/IoT binding

nanoFramework.Logging

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.3.0.10

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.841

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 6407ecb
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Dec 15 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.841
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  CatGate, 1.0.0.0
  System.Net, 1.10.38.33445
  nanoFramework.DependencyInjection, 1.0.12.27830
  nanoFramework.System.Text, 1.2.22.3995
  System.Device.Gpio, 1.1.22.60169
  nanoFramework.Hosting, 1.0.16.21277
  nanoFramework.Runtime.Native, 1.5.4.3
  nanoFramework.Json, 2.2.72.60323
  System.IO.Streams, 1.1.27.27650
  nanoFramework.M2Mqtt, 5.1.61.46161
  nanoFramework.Hardware.Esp32, 1.4.8.26232
  System.Text.RegularExpressions, 1.1.26.59508
  mscorlib, 1.12.0.4
  nanoFramework.System.Collections, 1.4.0.3
  System.Device.Wifi, 1.5.37.9881
  System.Threading, 1.1.8.6695
  nanoFramework.Runtime.Events, 1.11.1.42088
  nanoFramework.Logging, 1.1.52.20112

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

### Description

Curly brackets in the log message cause an exception.

Format error: wrong symbol at {}, column 19

### How to reproduce

Log a Json-String as Logmessage: 
Replace {} with () works as option.

```
        private void MqttService_ServiceChanged(object sender, EventArgs e)
        {
            string messageCommand = ((MqttServiceChangedEventArgs)e).Message;

            //string filteredLogMessage = new StringBuilder(messageCommand).Replace("{", "(").Replace("}", ")").ToString(); //Logger bug!
            //_logger.LogInformation($"Message received: {filteredLogMessage}");

            _logger.LogInformation($"Message received: {messageCommand}");
            AnalyzeCommands(messageCommand);
        }

```

### Expected behaviour

just log

### Screenshots

[image]

josesimoes, 2022-12-19:
Seems to be a matter of escaping the {} properly. Care to go for a PR with a fix? 😉

Ellerbach, 2022-12-20:
To be more precise, when logging, you should escae the `"{"` and `"}"` to `"\{"` and `"\}"`. That can be done there with a replace, why not. The reason is because this is using a string formater and it does uses those {} to replace what's inside!
And yes, we ❤️ PR :-)

DaveSchmid, 2022-12-20:
I hope I haven't forgotten anything. I'm doing this for the first time. It's all super complicated with Git and Github and everything in English.

josesimoes, 2022-12-20:
@DaveSchmid you're doing great! 🙂

closed 2023-01-04 by josesimoes