nanoframework/Home #1203: More than one device on SPI bus does not work

ToinoMau, 2023-01-09:
### Library/API/IoT binding

System.Device.Spi

### Visual Studio version

VS2022

### Target name(s)

ESP32 WROVER

### Description

SPI works as expected with one device connected to bus.
When a second device is declared and initialized on the same bus, both devices stop working even if the second is never called.
So, for the occurrence of this issue, is enough to declare and initialize the second device. 

### How to reproduce

//configure SPI bus-1
			Configuration.SetPinFunction(18, DeviceFunction.SPI1_CLOCK);//18
			Configuration.SetPinFunction(19, DeviceFunction.SPI1_MISO);//19
			Configuration.SetPinFunction(23, DeviceFunction.SPI1_MOSI);//23
			
			//device A
			SpiDevice spiDevice_A;
			SpiConnectionSettings connectionSettings_A;

			// Note: the ChipSelect pin should be adjusted to your device, here 4
			connectionSettings_A = new SpiConnectionSettings(1, 4);
			// You can adjust other settings as well in the connection
			connectionSettings_A.ClockFrequency = 100000;
			connectionSettings_A.DataBitLength = 8;
			connectionSettings_A.DataFlow = DataFlow.LsbFirst;
			connectionSettings_A.Mode = SpiMode.Mode3;

			// Then you create your SPI device by passing your settings
			spiDevice_A = SpiDevice.Create(connectionSettings_A);
			//..................

			//device B
			SpiDevice spiDevice_B;
			SpiConnectionSettings connectionSettings_B;
			Debug.WriteLine("Hello from sample for System.Device.Spi!");

			// Note: the ChipSelect pin should be adjusted to your device, here 13
			connectionSettings_B = new SpiConnectionSettings(1, 13);
			// You can adjust other settings as well in the connection
			connectionSettings_B.ClockFrequency = 100000;
			connectionSettings_B.DataBitLength = 8;
			connectionSettings_B.DataFlow = DataFlow.LsbFirst;
			connectionSettings_B.Mode = SpiMode.Mode3;

			// Then you create your SPI device by passing your settings
			spiDevice_B = SpiDevice.Create(connectionSettings_B);
			//..................

### Expected behaviour

On the exemplary project linked here (MultipleSPI_Test) two SpanByte are created and filled with a recognizable pattern and each are attributed to each SPI device and  transmitted for hardware signal observation in a loop.
The two SpanByte are very different apart in size:
                        // short SpanByte
			SpanByte shortSpanByte = new byte[2] { 42, 24 };
			//long SpanByte
			SpanByte longSpanByte = new byte[128];  
in order to produce two easily identifiable burst on the CLK signal and also in the CS pins.  

What is expected is a short CLK burst followed by a longer CLK burst corresponding to the sequenced transmission of the two devices.
What is also expected  is a short CS low state in pin-4 and a longer one in CS pin-13 each one in sync with the respective CLK.

What is observed is that only the short CLK burst appears on scope and the longer one is never observed.
Also it is CS-pin-13 that is fired. This pin  whoever belongs to the other device that transmits the longer burst. 
CS_pin-4 that belongs to the short observed burst is never fired.  

### Sample project or code

https://github.com/ToinoMau/MultipleSPI_Test

Ellerbach, 2023-01-22:
After some investigation:
* It seems that the bus is recreated somehow: https://github.com/nanoframework/nf-interpreter/blob/90151fe00b794a94238c51ec341a8255f5913001/targets/ESP32/_nanoCLR/System.Device.Spi/cpu_spi.cpp#L91
* The configuration returns then the last CS: https://github.com/nanoframework/nf-interpreter/blob/90151fe00b794a94238c51ec341a8255f5913001/targets/ESP32/_nanoCLR/System.Device.Spi/cpu_spi.cpp#L232
* So in short, you can only have 1 Chip Select per bus
* So far, if you want more chip select per device, you'll have to manually configure it, use a dummy CS for the SPI creation and then manual select both of your devices.

Ellerbach, 2023-01-23:
Few more on the investigation:
* Docs for SPI on IDF are here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#_CPPv4N29spi_device_interface_config_t12spics_io_numE
* While it seems each bus can handle 3 Chip Select at the same time, it seems that changing from one to another requires to adjust quite some elements on the bus
* Also STM32 is managed manually for the CS: https://github.com/nanoframework/nf-interpreter/blob/90151fe00b794a94238c51ec341a8255f5913001/targets/ChibiOS/_nanoCLR/System.Device.Spi/cpu_spi.cpp#L478

As a conclusion, we will have to adjust the ESP32 implementation to manage manually the Chip Select like done for the STM32

closed 2023-01-30 by josesimoes