nanoframework/Home #1208: Failed allocation for xxx blocks .. in nanoFramework.ResourceManager

DaveSchmid, 2023-01-14:
### Library/API/IoT binding

nanoFramework.ResourceManager

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.3.0.18

### Target name(s)

ESP32_REV0

### Firmware version

ESP32_REV0-1.8.0.877

### Device capabilities

ESP32 (ESP32-D0WD (revision 1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from ST (manufacturer 0x32 device 0x16406)
PSRAM: not available
Crystal 40MHz
MAC F0:08:D1:60:28:E4

### Description

Large resource files generate a "Failed allocation" error. Even larger files throw an OutOfMemoryException. But the real problem are the failed allocations. In my own (larger) project, even a favicon.ico (6 KB) is too big. Then a halved favicon appears in the browser.

### How to reproduce

I slightly modified the ManagedResources samples to generate the error:
It's checked in here:
https://github.com/DaveSchmid/Samples/tree/3b5dda821e3d6703d2226d136a4950a595ca39fc/samples/ManagedResources

### Screenshots

https://github.com/DaveSchmid/Samples/blob/3b5dda821e3d6703d2226d136a4950a595ca39fc/samples/ManagedResources/OutOfMemoryException.png

[image]

### Sample project or code

https://github.com/DaveSchmid/Samples/tree/3b5dda821e3d6703d2226d136a4950a595ca39fc/samples/ManagedResources

### Aditional information

The error occurs in the ResourceUtility class in the ResourceManager in the GetObject(rm, id) method, respectively externally.

The next method GetObject(rm, id, offset, length); would allow loading only part of the resource. Unfortunately, this is not used.

    public static class ResourceUtility
    {
        /// <summary>
        /// Gets the value of a specified <see cref="Object"/> resource for the current system culture.
        /// </summary>
        /// <param name="rm">The <see cref="ResourceManager"/> that contains the specified resources.</param>
        /// <param name="id">The integer identifier for the specified resource.</param>
        /// <returns>The value of the specified resource for the current system culture.</returns>
        [MethodImpl(MethodImplOptions.InternalCall)]
        extern public static object GetObject(System.Resources.ResourceManager rm, Enum id);

        /// <summary>
        /// Gets the value of a specified <see cref="Object"/> resource for the current system culture.
        /// </summary>
        /// <param name="rm">The <see cref="ResourceManager"/> that contains the specified resources.</param>
        /// <param name="id">The integer identifier for the specified resource.</param>
        /// <param name="offset">The offset for retrieving the value.</param>
        /// <param name="length">The length of the value to retrieve.</param>
        /// <returns>The value of the specified resource for the current system culture.</returns>
        [MethodImpl(MethodImplOptions.InternalCall)]
        extern public static object GetObject(System.Resources.ResourceManager rm, Enum id, int offset, int length);

alberk8, 2023-01-15:
@DaveSchmid , on a WROOM you only have about 70KB++ of RAM for nanoFramework,  so loading anything larger that 70KB will result in OOM.

DaveSchmid, 2023-01-15:
If the memory is used up, an OutOfMemoryException (OOM) exception is thrown. But with the "Failed allocation" no exception is thrown. The code just keeps running. But with incorrect data.
And also it should be able to stream a larger file. Flash is enough available.

(What does the WR in WROOM actually stand for? :-)

alberk8, 2023-01-15:
I don't know the details how nanoCLR works but what I know with the failed allocation is a form of OOM in nF.  I get this when create a long hashtable.  I am pretty sure hashtable does not run on flash (XIP) as it is dynamically created.  

ESP32 WROOM is the marketing name for an ESP32 module that does not have extra PSRAM (pseudo SRAM).  the WROVER module has up to 4MB extra for nF to use.

DaveSchmid, 2023-01-15:
That is interesting. I have also seen this failed allocation before in another context. But I can't remember where.

josesimoes, 2023-01-16:
The "failed allocation message" is a informational mesage from the execution engine to report that it has (no surprise here) failed to allocate N blocks of memory. When this message is shown nothing wrong necessarily happens with the application. 
After this information the GC is forced to run and will try to reclaim memory from unused objects to the managed heap.
After this, the requested allocation process will take place once more. If it succeds the execution will continue happilly. If it fails then you'll see the OOM exception.

josesimoes, 2023-01-16:
Regarding the suggestion to stream objects by using `GetObject` that's not the way the resource manager is coded. Resources are loaded in memory allocated from managed heap.

I understand your suggestion and it makes sense to me that this could be used as an alternative from certain resource types, like the icon as you describe above, and that would be streamed when requested. For that to happen a serious rewrite on resource manager would have to happen. Both C# and native. Mostly on the latter.

DaveSchmid, 2023-01-17:
ok, I analyzed the cut off favicon.ico again. I admit it was my fault. I got confused by the "failed allocation message". In fact, the reason for the truncated images in the browser is a bug in the web server.

The method `WebServer.SendFileOverHTTP(HttpListenerResponse response, string fileName, byte[] content, string contentType = "")` cuts a byte array into small pieces to send (what is not working properly). The favicon used in Webserver.Sample is so small that it is sent in the first pass. Apparently nobody noticed this.

So I'm closing here.

closed 2023-01-17 by DaveSchmid

Ellerbach, 2023-01-19:
> The method `WebServer.SendFileOverHTTP(HttpListenerResponse response, string fileName, byte[] content, string contentType = "")` cuts a byte array into small pieces to send (what is not working [...]

Do you mind explaining more in details what's not working with this function? I works well to send large files for example, so not sure why it would not properly work for an icon.

DaveSchmid, 2023-01-19:
> Do you mind explaining more in details what's not working with this function? I works well to send large files for example, so not sure why it would not properly work for an icon.

yes there is a problem. In fact, I've found also many others. And I'm not sure if you want to know all of them. I would be a good tester. But you would hate me after a while. And that would take a lot of time for you and for me. I'm not sure if I have enough patience. Should I create an issue for each bug I see? Maybe some are not errors at all maybe problems in my setup or misunderstandings or misinterpretations of how it should be.

josesimoes, 2023-01-19:
@DaveSchmid everyone here gives it's best. It's based on both personal and community effort. All that you see here (and that you are using) comes from that.
Being a developer yourself I'm sure you'll appreciate that _sometimes_ bugs slip into production. No one releases buggy code on purpose and, needless to say, that in nanoFramework we have no interest on having libraries and samples that do not work or have errors.

Sooo...
 
1. Yes, we are definitely interested in knowing **all** the issues you've found so they can be corrected. As a side note: it can't be _thaaat_ bad, as the library is around for quite some time and has a fair number of downloads.
2. If some of those are not issues, but rather that you are misusing the API, we'll gladly give advice on how to it better.
3. If something can be improved we'll gladly hear any suggestions towards that. And we love PRs!
4. As you've experienced by now, we have a very friendly community here. For sure no one will hate you for reporting bugs and /or performing thorough and exhaustive testing of the libraries and firmware.
5. Time is on short supply for everyone involved. You included, I'm sure. We appreciate and thank you for all the time you want to invest on the project. We'll do our best to keep up with you for as long as you want.
6. On how to handle this: better to start discussing whatever you're finding in Discord and then issues can be raised for actual issues.

DaveSchmid, 2023-01-19:
> everyone here gives it's best. All that you see here (and that you are using) comes from that and it's based on both personal and community effort. Being a developer yourself I'm sure you'll appreciate that _sometimes_ bugs slip into production. No one releases buggy code on purpose and, needless to say, that in nanoFramework we have no interest on having libraries and samples that do not work or have errors.

I have great respect for what you have done and are doing. II wouldn't dare to do that.  

I have been a programmer for many years and know that you make many more mistakes than you self think. And it's not always easy to admit it. Specially when someone else points his finger at it. That's why I normally prefer to keep my code to myself. But I will do my best to help.

> 1. Yes, we are definitely interested in knowing **all** the issues you've found so they can be corrected. As a side note: it can't be _thaaat_ bad, as the library is around for quite some time and has a fair number of downloads.

I know that this library has been around for a long time. I tried to use it many years ago and then gave it up because it was just too buggy. So you should not be too irritated by the number of downloads. I think there are a lot of users who just want to try something out without ever giving feedback. And extremely many who also have problems with the English language (like me, for example) and just try another way to solve their Requirements. They won't get in touch with you either. That's not meant to be a criticism, quite the opposite. I was amazed to see a readme in Chinese. I think that's super cool!

> 6. On how to handle this: better to start discussing whatever you're finding in Discord and then issues can be raised for actual issues.

I think the Discord Chat is very commendable. I also think that this is really good to discuss solutions or give tips. Especially if you are fast to read and write with English. Otherwise it is rather difficult. Personally, however, I have a lot of trouble with a chat as a knowledge store. It sometimes seems to me like a big pot of alphabet soup. For example, I searched the internet for "Failed allocation for ... Block.. in nanoFramework". If you search for it now, you will find exactly this discussion here and it explains that it is not a error its a feature. So I think an issue here is better even if it is not one in reality.

josesimoes, 2023-01-20:
@DaveSchmid no problem following-up this on GitHub. When you want to tackle this, please start a [Discussion](https://github.com/nanoframework/Home/discussions) and we'll take it from there.
(kindly ping me and @Ellerbach when you do that)

sky9mike, 2023-07-30:
> The "failed allocation message" is a informational mesage from the execution engine to report that it has (no surprise here) failed to allocate N blocks of memory. When this message is shown nothing wrong necessarily happens with the application.

I came her via search @josesimoes. I humbly suggesting this message should be improved to not freak out devs.

Example:

**Note:** Failed allocation for 615 blocks, 7380 bytes **(this is not a real issue)**

I would go further and say this should not even be displayed and, if is desirable for hard core nF devs, a static property control this.

josesimoes, 2023-07-31:
@sky9mike understood and I agree with that. We'll change this to show only when verbose option is selected for GC output.