nanoframework/Home #1210: GpioChangeCounter

yakeer, 2023-01-18:
### Description

Please add GpioChangeCounter functionality.

It's the only counter that uses internal ESP32 counter, which supports ultra fast counting (I personally use it in Encoder counters).
https://docs.nanoframework.net/api/Windows.Devices.Gpio.GpioChangeCounter.html

### How to solve the problem

Create support for deprecated functionality.

Ellerbach, 2023-03-23:
I will investigate how to best add this feature and fir into the System.Device.Gpio implementation.

Ellerbach, 2023-03-24:
@yakeer you will be happy, feature added and as a bonus, much better than before. You now have native support for 2 pins, so directly what you need for a rotary encoder.
Feedbacks welcome! And enjoin!

yakeer, 2023-03-24:
Thanks, as soon as it's in the framework I will test it with my firmware.

Ellerbach, 2023-03-25:
> Thanks, as soon as it's in the framework I will test it with my firmware.

All merged. You'll have to flash your device with the latest firmware. The Pulse Count is now part of nanoFramewrok.Hardware.Esp32 nuget. And it went thru a rename. Let me know!

closed 2023-03-25 by Ellerbach