nanoframework/Home #1211: SerialPort.Read does not return 0 (zero) values in buffer (STM32F769I)

Sprocket1979, 2023-01-20:
### Library/API/IoT binding

System.IO.Ports

### Target name(s)

STM32F769I_DISCO

### Firmware version

1.8.0.806

### Description

When sending a telegram from a KNX TpUart device to Serialport, values 0 are being selectively removed from the buffer.

I send a telegram like this to the COM port.
byte[] buf = new byte[] { 0x68, 0x00, 0x00, 0x68, 0xD3, 0xC1, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0xA1, 0x16 }; 

SerialPort.Read buffer receive this. DATA = 68-68-D3-C1-01-01-01-01-01-A1-16
The 0 values are not i the buffer.

### How to reproduce

        public static void InitKnxTpUart()
        {
            try
            {

                _serialDevice = new SerialPort("COM6");
                _serialDevice.Mode = SerialMode.Normal;
                _serialDevice.BaudRate = 19200;
                _serialDevice.Parity = Parity.Even;
                _serialDevice.StopBits = StopBits.One;
                _serialDevice.DataBits = 8;

                _serialDevice.DataReceived += _serialDevice_DataReceived;

                _serialDevice.Open();

                IsOpen = true;

            }
            catch
            {
                IsOpen = false;
            }
        }

        private static void _serialDevice_DataReceived(object sender, SerialDataReceivedEventArgs e)
        {
            try
            {

                var serialPort = (SerialPort)sender;
                byte[] buffer = new byte[serialPort.BytesToRead];
                serialPort.Read(buffer, 0, buffer.Length);

            }
            catch
            {
            }

        }

### Screenshots

[image]

Mayuresh0072, 2023-01-22:
@josesimoes @DaveSchmid @martijns Hello, I will work on this issue, Please assign me this issue. Thanks.

josesimoes, 2023-02-23:
@Mayuresh0072 any progress on this please?

Mayuresh0072, 2023-03-01:
I think We have to change DataReceived Method as follows, to wait until all the expected bytes have been received before processing the data.
Please check and tell me if it is a correct solution.
private static void _serialDevice_DataReceived(object sender, SerialDataReceivedEventArgs e)
{
    try
    {
        var serialPort = (SerialPort)sender;
        int bytesToRead = serialPort.BytesToRead;
        byte[] buffer = new byte[bytesToRead];
        int bytesRead = 0;

        while (bytesRead < bytesToRead)
        {
            int numBytes = serialPort.Read(buffer, bytesRead, bytesToRead - bytesRead);
            if (numBytes == 0)
            {
                // If no bytes are read, wait a little and try again
                Thread.Sleep(10);
            }
            else
            {
                bytesRead += numBytes;
            }
        }

        // Now process the received data in the buffer
        // ...

    }
    catch (Exception ex)
    {
        // Handle exceptions
        Console.WriteLine(ex.Message);
    }
}
I think We have to change DataReceived Method as follows, to wait until all the expected bytes have been received before processing the data

Sprocket1979, 2023-03-01:
I have tried to read 1 byte at a time instead of the DataReceived method with the same result.
 
 private static void RxThread()
        {
            byte[] buffer = new byte[1];
            while (true)
            {
                if (serialPort.Read(buffer, 0, 1) > 0)
                {
                    knxDataFrame.Add(buffer[0]);
                }
            }
        }

Sprocket1979, 2023-04-04:
Any news ?
Thanks.

networkfusion, 2023-04-20:
This (might) have been fixed upstream (ChibiOS), but to add extra detail, here is the test I am using:

```
using System;
using System.Diagnostics;
using System.Threading;
using System.IO.Ports;

namespace nano_stm32f769i_serial_tests
{
    public class Program
    {
        private static SerialPort _serialDevice;

        public static void Main()
        {
            Debug.WriteLine("nF ComPort6 test");
            // This is a loopback test for COM6
            // make sure that RX and TX are joined...
            // connecter CN13
            // pins: D1 to D0
            //
            if (InitComPort6())
            {
                Debug.WriteLine("COM6 initialized.");
            }
            else 
            {
                Debug.WriteLine("COM6 initialization failed!.");
            }

            // TODO: Send some data!
            byte[] sendBuffer = new byte[] { 0x68, 0x00, 0x00, 0x68, 0xD3, 0xC1, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0xA1, 0x16 };
            _serialDevice.Write(sendBuffer, 0, sendBuffer.Length);

            //Thread.Sleep(2000);
            // TODO: check read against sendBuffer!

            Thread.Sleep(Timeout.Infinite);
        }

        public static bool InitComPort6()
        {
            try
            {

                _serialDevice = new SerialPort("COM6")
                {
                    Mode = SerialMode.Normal,
                    BaudRate = 19200,
                    Parity = Parity.Even,
                    StopBits = StopBits.One,
                    DataBits = 8
                };

                _serialDevice.DataReceived += SerialDevice_C6_DataReceived;

                _serialDevice.Open();

                return true;

            }
            catch
            {
                return false;
            }
        }

        private static void SerialDevice_C6_DataReceived(object sender, SerialDataReceivedEventArgs e)
        {
            try
            {
                var serialPort = (SerialPort)sender;
                Debug.WriteLine($"Received data from: {serialPort.PortName}");
                byte[] buffer = new byte[serialPort.BytesToRead];
                var bytesRead = serialPort.Read(buffer, 0, buffer.Length);
                if (bytesRead > 0)
                {
                    Debug.WriteLine($"Bytes Read = {bytesRead}");
                    Debug.WriteLine($"Content as hex:");
                    foreach (var b in buffer)
                    {
                        Debug.Write(b.ToString("X2")); // print hex output!
                    }
                    Debug.WriteLine();
                }

            }
            catch (Exception ex)
            {
                Debug.WriteLine($"Error receiving data! {ex}");
            }

        }
    }
}
```

With the output:
[image]

I am worried about the extra received events, but not sure if that is part of this issue.

If someone can improve the sample above to help further debug, it would be very helpful.

Sprocket1979, 2023-04-24:
@networkfusion Loopback worked for me as well. I think the issue is only when using an external device. I tested my KnxUart device on Esp32, and that worked.

networkfusion, 2023-04-24:
Okay, thanks, I have managed to repro.

```
using System;
using System.Diagnostics;
using System.Threading;
using System.IO.Ports;

namespace nano_stm32f769i_serial_tests
{
    public class Program
    {
        private static SerialPort _serialDevice_Com5;
        private static SerialPort _serialDevice_Com6;

        public static void Main()
        {
            Debug.WriteLine("nF ComPort6 test");
            // This is a test for COM5 to COM6

            // connecter: CN13 (Arduino Connector)
            // pin: D0
            // to:
            // connector: CN2 (WiFi daughter board connector)
            // pin: 1

            // connecter: CN13 (Arduino Connector)
            // pin: D1
            // to:
            // connector: CN2 (WiFi daughter board connector)
            // pin: 8
            //
            if (InitComPorts())
            {
                Debug.WriteLine("COM5 and COM6 initialized.");
            }
            else
            {
                Debug.WriteLine("COM port initialization failed!.");
            }

            // Test send bytes
            byte[] sendBuffer = new byte[] { 0x68, 0x00, 0x00, 0x68, 0xD3, 0xC1, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0xA1, 0x16 };

            while (true)
            {
                var outputString = "";
                Debug.WriteLine($"Sending some bytes:");
                Debug.WriteLine("  (debug output as hex)");
                foreach (var b in sendBuffer)
                {
                    outputString += $"0x{b.ToString("X2")},"; // print hex output!
                }
                Debug.WriteLine(outputString.TrimEnd(','));
                _serialDevice_Com6.Write(sendBuffer, 0, sendBuffer.Length);

                Thread.Sleep(5000);

            }

            
        }

        public static bool InitComPorts()
        {
            try
            {
                _serialDevice_Com5 = new SerialPort("COM5")
                {
                    Mode = SerialMode.Normal,
                    BaudRate = 19200,
                    Parity = Parity.Even,
                    StopBits = StopBits.One,
                    DataBits = 8
                };
                _serialDevice_Com5.DataReceived += SerialDevice_Com_DataReceived;
                _serialDevice_Com5.Open();

                _serialDevice_Com6 = new SerialPort("COM6")
                {
                    Mode = SerialMode.Normal,
                    BaudRate = 19200,
                    Parity = Parity.Even,
                    StopBits = StopBits.One,
                    DataBits = 8
                };
                _serialDevice_Com6.DataReceived += SerialDevice_Com_DataReceived;
                _serialDevice_Com6.Open();

                return true;

            }
            catch
            {
                return false;
            }
        }

        private static void SerialDevice_Com_DataReceived(object sender, SerialDataReceivedEventArgs e)
        {
            var serialPort = (SerialPort)sender;
            Debug.WriteLine($"[Start of {serialPort.PortName} Event!]");
            try
            {
                Debug.WriteLine($"Received data from: {serialPort.PortName}");
                byte[] buffer = new byte[serialPort.BytesToRead];
                var bytesRead = serialPort.Read(buffer, 0, buffer.Length);
                if (bytesRead > 0)
                {
                    Debug.WriteLine($"Bytes Read = {bytesRead}");
                    Debug.WriteLine("  (debug output as hex)");
                    var outputString = "";
                    foreach (var b in buffer)
                    {
                        outputString += $"0x{ b.ToString("X2") },"; // print hex output!
                    }
                    Debug.WriteLine(outputString.TrimEnd(','));
                }

            }
            catch (Exception ex)
            {
                Debug.WriteLine($"Error receiving data! {ex}");
            }
            Debug.WriteLine($"[End of {serialPort.PortName} Event!]");

        }
    }
}

```

networkfusion, 2023-04-26:
After some more tests, I beleive this is linked to #1270 and specifically related to the one wire lib.
A work around (if not required) is to turn off one wire (`"API_nanoFramework.Device.OneWire": "OFF",` in https://github.com/nanoframework/nf-interpreter/blob/e3864c2cf10a5576a88bc97d8a521a71a763d2aa/targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakePresets.json#L40

If someone else can test, it would be helpful.

josesimoes, 2023-11-30:
@Sprocket1979 can you please retest this?
I think it may have been fixed with nanoframework/nf-interpreter#2834.