nanoframework/Home #1212: Visual Studio crashes and later shows an error message

DaveSchmid, 2023-01-21:
### Tool

Visual Studio extension

### Description

Visual Studio (2022 and 2019) crashes unexpectedly and shows an error message to disable the extension after automatic restart.

### How to reproduce

Happens unexpectedly during non-specific actions.

### Expected behaviour

Should be solid as iron!

### Screenshots

[image]

[image]

### Aditional context

Microsoft Visual Studio Community 2022 (64-bit) - Current Version 17.4.4
nanoFramework extension v2022.3.0.18

Ellerbach, 2023-01-21:
Can you please give a bit more context? What kind of project do you have open? How do you close VS? What are you doing in general when this happens?
I have the same extension version installed on a Microsoft Visual Studio Enterprise 2022 (64-bit) - Current Version 17.4.4 and I did not see any of the issue you describe. I have Visual Studio open for very different kind of projects and did not experience any crashes with this version or any of the previous ones for years now!

DaveSchmid, 2023-01-24:
It just happens out of any context.
I have now activated this Fusion logging. 
https://stackoverflow.com/questions/255669/how-to-enable-assembly-bind-failure-logging-fusion-in-net 
But I should now know which the GUID is of the extension?

DaveSchmid, 2023-01-24:
but I don't think that's the correct log anyway.

josesimoes, 2023-01-24:
I've seen this happen randomly. To this day I have no idea what could be causing it. Even when debugging the extension this happens and there is no exception thrown in VS. I suspect that's coming from the debugger library and/or from the debugger extension. 

If you can find any clue on the root cause, let's look into this. Otherwise... 
Personally I don't think this is impacting the performance or the way the extension works, so I wouldn't put much effort on this.

DaveSchmid, 2023-01-24:
I absolutely agree with you.
It's just not pretty.

josesimoes, 2023-01-24:
And I can only agree with you! 😅

josesimoes, 2024-03-25:
If you don't have details, I'll close the issue and it can be open at a later time when more details are available.
Because if we can't reproduce it, no way to even start looking on what could be causing it...

closed 2024-03-25 by josesimoes