nanoframework/Home #1213: SSL Request to nanoFramework.WebServer throws a System.Net.Sockets.SocketException

DaveSchmid, 2023-01-21:
### Library/API/IoT binding

nanoFramework.WebServer

### Visual Studio version

VS2022 17.4.4

### .NET nanoFramework extension version

v2022.3.0.18

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.877

### Device capabilities

ESP32 (ESP32-D0WDQ6 (revision 1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from ST (manufacturer 0x32 device 0x16406)
PSRAM: undetermined
Crystal 40MHz
MAC 24:6F:28:9E:31:30

### Description

An SSL request in the WebServer.Sample throws a System.Net.Sockets.SocketException in the background.
However, the code continues to run. Everything looks good. Fault?

Ready.
The nanoDevice runtime is loading the application assemblies and starting execution.
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (4) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
    ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
The thread '<No Name>' (0x5) has exited with code 0 (0x0).
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (4) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
    ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
The thread '<No Name>' (0x7) has exited with code 0 (0x0).

### How to reproduce

The error is visible with the Samples\samples\Webserver\WebServer.Sample Project with every web request in the debugging console. 

- Get and start the Example: https://github.com/nanoframework/Samples/tree/main/samples/Webserver/WebServer.Sample
- uncomment line8 in Program.cs:  #define USE_TLS
- set ssid and password (line 36/37)
- load https://device-ip/favicon.ico in a Browser

### Expected behaviour

No exception in the debug console

### Sample project or code

https://github.com/nanoframework/Samples/tree/main/samples/Webserver/WebServer.Sample

Ellerbach, 2023-01-22:
I guess this is coming from the certificate used which is most likely out dated. What if you are using your own certificate?

DaveSchmid, 2023-01-24:
ok, I tested it with valid certificate. 
Unfortunately, the exception still comes.

[image]

nh43de, 2023-11-18:
I am also having this issue - 

Program starting...
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (5) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::poll [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::Poll [IP: 0011] ++++
    ++++ System.Net.Sockets.Socket::Accept [IP: 0017] ++++
    ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 0022] ++++
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (5) ++++

alberk8, 2023-11-19:
Try to verify the TLS with openssl, you can do this on a Linux or WSL running Linux.  For the example below I am running against mosquitto.org mqtt server which shows that the Cert is invalid.  

openssl s_client -connect www.exampleurl.com:443 -showcerts

[image]

networkfusion, 2023-11-19:
Noting... I am currently getting something like this after the latest changes to system.io.file nuget changes. I suspect it will sort itself after some nuget auto updates.

CoryCharlton, 2023-11-28:
> I am also having this issue with my own certificate
> 
> Program starting... ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (5) ++++ ++++ Message: ++++ System.Net.Sockets.NativeSocket::poll [IP: 0000] ++++ ++++ System.Net.Sockets.Socket::Poll [IP: 0011] ++++ ++++ System.Net.Sockets.Socket::Accept [IP: 0017] ++++ ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 0022] ++++ ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (5) ++++
> 
> But I was able to resolve by adding values to the cert fields on creation (before I was leaving them blank)

Filling out all the fields does resolve this `CLR_E_FAIL (5)` error that I was getting in `SecureRead`. However the `CLR_E_FAIL (4)` error in `SecureAccept` persists:

```
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (4) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
    ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
```

Unfortunately I haven't been able to determine what HRESULT 4 and 5 are indicating in this context.

CoryCharlton, 2023-11-28:
Also of interest is that when I use `openssl s_client -connect 10.0.10.118:443 -showcerts` the HRESULT from `SecureRead` just increments (which suggests to me that it doesn't mean anything useful :():

```
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (9) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureRead [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Read [IP: 0040] ++++
    ++++ System.Net.InputNetworkStreamWrapper::RefillInternalBuffer [IP: 0038] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
    ++++ System.Net.HttpListenerContext::get_Request [IP: 000d] ++++
The thread '<No Name>' (0x9) has exited with code 0 (0x0).
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (10) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureRead [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Read [IP: 0040] ++++
    ++++ System.Net.InputNetworkStreamWrapper::RefillInternalBuffer [IP: 0038] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
    ++++ System.Net.HttpListenerContext::get_Request [IP: 000d] ++++
The thread '<No Name>' (0xa) has exited with code 0 (0x0).
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (11) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureRead [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Read [IP: 0040] ++++
    ++++ System.Net.InputNetworkStreamWrapper::RefillInternalBuffer [IP: 0038] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
    ++++ System.Net.HttpListenerContext::get_Request [IP: 000d] ++++
The thread '<No Name>' (0xb) has exited with code 0 (0x0).
```

Edit: I just noticed in this case the HRESULT corresponds to the thread id...

CoryCharlton, 2023-11-29:
I was misreading that output and even though the error code is set on the SocketException it is not included in that output. I added some addition debug logging and for me the error code in `SecureAccept` is always:

```
#define MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE   -0x7780

A fatal alert message was received from our peer.
```

That being said everything works eventually so 🤷

networkfusion, 2024-05-05:
Any further comment on this as no further reported issues?

josesimoes, 2024-09-24:
Following the confirmation above with Wireshark capture and no further feedback from the original poster, I'm closing this.
Feel free to reopen if needed.

closed 2024-09-24 by josesimoes