nanoframework/Home #1216: WebServer.SendFileOverHTTP truncates Files

DaveSchmid, 2023-01-23:
### Library/API/IoT binding

nanoFramework.WebServer

### Visual Studio version

VS2022

### .NET nanoFramework extension version

v2022.3.0.18

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.877

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 6407ecb
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Jan  6 2023
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.877
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  WebServer.Sample, 1.0.0.0
  nanoFramework.System.Text, 1.2.37.0
  mscorlib, 1.14.3.0
  System.IO.FileSystem, 1.1.23.0
  Windows.Storage, 1.5.33.0
  System.Device.Gpio, 1.1.28.0
  nanoFramework.ResourceManager, 1.2.13.0
  nanoFramework.System.Collections, 1.5.18.0
  System.Device.Wifi, 1.5.51.0
  System.IO.Streams, 1.1.38.0
  nanoFramework.Runtime.Events, 1.11.6.0
  System.Net, 1.10.52.0
  System.Net.Http, 1.5.78.0
  Windows.Storage.Streams, 1.14.24.46541
  System.Threading, 1.1.19.33722
  nanoFramework.WebServer, 1.1.0.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

The WebServer.SendFileOverHTTP method only returns the first 1KB for larger resources to the Browser.

In Chrome, the image is displayed in the correct size, but only the top part of the image.

[image]

[image]

In Firefox, the image is not displayed and an error message is displayed: Image corrupt or truncated.

[image]

Further network investigation has shown that the error is also visible there.
See picture and attached Wireshark pcap file.

[image]

[sniff.pcap.zip](https://github.com/nanoframework/Home/files/10479518/sniff.pcap.zip)

In the Http stream you can see that the same data blocks are sent over and over again (marked dark blue):

[image]

The error is very easy to solve. But I'm not sure if my variant works for different device types. I can only test it with ESP32.

In the WebServer.SendFileOverHTTP methods, files and resources are broken down into 1KB chunks. This seems unnecessary and generates the error. The HttpListenerResponse.ContentLength64 expects a long but OutputStream.Write can only accept Int32. But that's not really a problem:
Int32.MaxValue = The value of this constant is 2,147,483,647
2147483647 Bytes = 2 Gigabytes

What was the idea for the 1 KB pieces?

### How to reproduce

Modified WebServer.Sample with modified SendFileOverHTTP from WebServer.cs :
https://github.com/DaveSchmid/Samples/releases/tag/WebServer.Sample-Errors
Sample shows misbehave of truncated Files.

1. Start modified WebServer.Sample
2. open request in Chrome:
     https://yourdeviceip/favicon2.png
     https://yourdeviceip/favicon3.png

### Expected behaviour

The web server should deliver entire files.

### Screenshots

The whole file should look like this:

[image]

### Sample project or code

Modified WebServer.Sample with modified SendFileOverHTTP from WebServer.cs :
https://github.com/DaveSchmid/Samples/releases/tag/WebServer.Sample-Errors
Sample shows misbehave of truncated Files.

Ellerbach, 2023-01-23:
> What was the idea for the 1 KB pieces?

Yes, that's needed because devices are constraints. This has been proven to work nicely without any issue. So not sure what is the core issue you are having. I can't reproduce it and I've been using it for years now with large files and it just works fine on my end.

DaveSchmid, 2023-01-23:
Perhaps something has changed in System.Net or in the underlying OS? It also seems strange to me that you just call response.OutputStream.Write multiple times without saying when to send the buffer. (for example with a flush or something..). If you look at the description of OutputStream then it also says:  The first write sends ...

        //
        // Summary:
        //     Gets a Stream object to which a response can be written.
        //
        // Value:
        //     A Stream object to which a response can be written.
        //
        // Remarks:
        //     The first write to the output stream sends a response to the client.
        public Stream OutputStream

DaveSchmid, 2023-01-23:
But unfortunately that doesn't work either:

            response.SendChunked = true;

            //Now loop to send all the data.
            for (long bytesSent = 0; bytesSent < content.Length;)
            {
                // Determines amount of data left
                long bytesToSend = content.Length - bytesSent;
                bytesToSend = bytesToSend < MaxSizeBuffer ? bytesToSend : MaxSizeBuffer;

                // Writes data to output stream
                response.OutputStream.Write(content, (int)bytesSent, (int)bytesToSend);

                // allow some time to physically send the bits. Can be reduce to 10 or even less if not too much other code running in parallel

                // update bytes sent
                bytesSent += bytesToSend;
            }
            response.OutputStream.Close();

DaveSchmid, 2023-01-23:
So I took a look at nanoFramework.System.Net.Http\Http\System.Net._OutputNetworkStreamWrapper.cs. The public override void Write(byte[] buffer, int offset, int size) method on line 270 is the entry point for the request. I compared response.SendChunked = true and response.SendChunked = false in the network analyzer. It does exactly what you want:

Wireshark request unchunked:
[image]

The parsed HTTP stream (unchunked):
[image]

And now the variant chunked:
[image]

[image]

This is exactly the function you want.
But the call stays the same except the property SendChunked has to be set.

DaveSchmid, 2023-01-23:
It is not necessary to call close(). And the slicing is not necessary either, or rather wrong.

I made another commit: https://github.com/nanoframework/nanoFramework.WebServer/pull/215/commits/b23a6a74da3903c014acb4f951aa9f4fc86b783e

Ellerbach, 2023-01-24:
> Perhaps something has changed in System.Net or in the underlying OS? It also seems strange to me that you just call response.OutputStream.Write multiple times without saying when to send the buffer. [...]

Yes, most likely. I'll rebuild a new image and will test with large files. And thanks for the investigation. Still, you can't allocate large buffers, that won't work on small devices and you'll still have to go for the chunk pattern.

DaveSchmid, 2023-01-24:
@Ellerbach Yes please take a close look at this issue. Even with different devices. Unfortunately I only have one type. Maybe it would be good if there were different variants of the call? For different devices. 
In any case, I'm in no hurry, it's more important to test this problem accurately.
And anyway, the thanks are yours

josesimoes, 2023-01-24:
System.Net hasn't changed in a while. Nor has lwIP which is the network stack used by both ESP32 and STM32. Unless Espressif has made changes on their lwIP wrappers... 

Chunked is the way to go. As a file is read by FileIO, there is already memory allocation for the complete file size. Creating another buffer with the size of the file, will consume _another_ chunk of memory just to hold a copy of the same data. Therefore the chunked approach.

Regarding the `SendChunked` property, nice finding! That's most likely the key for all this.

josesimoes, 2023-01-24:
> It is not necessary to call close(). 

Why is that? If you don't close the socket it will be lingering and taking up one connection and one socket that would be free for reuse otherwise.

Ellerbach, 2023-01-24:
@DaveSchmid I can reproduce the behavior. Got a partial file as well with latest firmware, latest nugets, latest everything.
I'm using a very basic ESP32 with no PSRAM, so can't allocate all the memory for the file. So it will be perfect case to make it working.
So trying to find out the core root issue and yes, will look at the SendChunked!

Ellerbach, 2023-01-24:
@DaveSchmid found the root cause issues. There were 2. One in System.Net.Http and the other one in the WebServer.
All up, it's working with or without the chunked set. I'll add the chunk set in the webserver so it's a bit cleaner but technically not needed as we always specify the file size. If we would not, then chunk should be helpful.
And thanks again for the investigation! Really helped.

Ellerbach, 2023-01-24:
@DaveSchmid give it a bit of time to be updated and merged for both PR and the nuget. Thanks again for the investigation. What I was using was much old and before the System.IO.Stream... So it's the reason it was still working!

closed 2023-01-24 by Ellerbach

Ellerbach, 2023-01-24:
> The socket is already closed. It throws an exception.

which line exactly?

DaveSchmid, 2023-01-24:
@Ellerbach in my example above. The last line.
https://github.com/nanoframework/Home/issues/1216#issuecomment-1400880828