nanoframework/Home #1217: nanoFramework.UI.Bitmap missing

DaveSchmid, 2023-01-24:
### Tool

Visual Studio extension

### Description

Adding Images of the type .bmp , .gif and .jpg as resources make the VS solution uncompilable.

Error	CS0234	The type or namespace name 'UI' does not exist in the namespace 'nanoFramework' (are you missing an assembly reference?):

[image]

[image]

If you play around and practice long enough with deleting and different kinds of adding another error message appears:

[image]

The resource System.Drawing was automatically added to the project. Compiling is only possible again once this has been deleted.

### How to reproduce

1. I am creating a new Blank Application Solution.
2. add nanoFramework.ResourceManager as NuGet Package 
3. add a new Resource File (.NET nanoFramework) to the project
4. add some Files (Images: .png, .pcx, .tif and Sound: .wav) to the Resource1.resx (per drag and drop or manuell,  makes no difference). The pcx is not recognized as an image (works fine anyway)
6. I compile the project. No problem everything perfect.
7. Now I add more pictures (.bmp, .gif, .jpg)
8. Now the project can no longer be compiled! 

My example (before and after the error):

[NFImageResscources_working.zip](https://github.com/nanoframework/Home/files/10491163/NFImageResscources_working.zip)
[NFImageResscources_notworking.zip](https://github.com/nanoframework/Home/files/10491165/NFImageResscources_notworking.zip)

### Expected behaviour

1. it would be nice if adding a Resource.resx automatically adds the nanoFramework.ResourceManager reference.
2. nanoFramework.UI.Bitmap should be available. Or images should be added as normal files (BinaryResources instead of BitmapResources). Nobody wants to edit images in VS, anyway!

### Aditional context

Microsoft Visual Studio Community 2022 (64-bit) - Current Version 17.4.4
.NET nanoFramework VS Extension v2022.3.0.18

josesimoes, 2023-01-24:
Duplicated of #844 

This is the standard behavior. Unfortunately we can't do much about it. This depends on the VS extensibility library and there is no option to extend resources to other languages.

Reference to ResourceManager nuget has to be added manually.
Reference to UI has to be added manually if images are being used.

No native support for pcx, so that's handled as a binary resource.

DaveSchmid, 2023-01-24:
> Duplicated of #844
>
> This is the standard behavior. Unfortunately we can't do much about it. This depends on the VS extensibility library and there is no option to extend resources to other [...]

No the problem is not the manual adding. The problem is the fact that nanoFramework.UI.Bitmap does not exist in nanoFramework.System.Drawing.

josesimoes, 2023-01-24:
I got lost on your explanation. I see... that's probably in the resource processing task....

TerryFogg, 2023-06-02:
I loaded up your [NFImageResscources_working.zip]
In this project I didn't see a reference to the graphics library.
Bitmaps types and namespaces are supported by the graphics nuget not by the resource manager.
You need both resource manager and graphics references.

Does this answer your question?

DaveSchmid, 2023-06-02:
no, the whole thing is not a question, it is the explanation of what exactly is missing. 
Someone has to write the code for nanoFramework.UI.Bitmap. Without this code .bmp, .gif, .jpg will not work.

DaveSchmid, 2023-06-02:
Maybe it can just be a dummy class. Or it is very difficult. I don't know it.

TerryFogg, 2023-06-02:
The code for nanoFramework.UI.Bitmap. is already written.

nanoFramework.Graphics/Bitmap.cs at main · nanoframework/nanoFramework.Graphics · GitHub<https://github.com/nanoframework/nanoFramework.Graphics/blob/main/nanoFramework.Graphics/Primitive/Bitmap.cs>

DaveSchmid, 2023-06-02:
Ok, with the nuget nanoFramework.Graphics I can now compile but unfortunately not upload it to the device. This message appears: [The connected target does not have support for nanoFramework.Graphics.]
But I don't want to edit the graphic on the device, I just want to upload the resource to the device. I actually only need the part of nanoFramework.Graphics for the ResourceManager in VisualStudio.

DaveSchmid, 2023-06-02:
ok, that's cool but i don't want any graphic support on the device, i just want to upload a jpg for the web server.

TerryFogg, 2023-06-02:
You can do the following I think, not near a computer at the moment.
Change the extension of your graphics to .dat or something, the software automatically adds template code based on extension type, .dat should not need graphics reference.

DaveSchmid, 2023-06-02:
Unfortunately, this workaround doesn't work either. It seems that the VisualStudio ResourceManager dialog parses the files somehow. He notices that it is a graphics file. No matter what file extension.

DaveSchmid, 2023-06-02:
oh, i found a way! It works with .txt Text File.  
The whole thing is a bit ugly but this way it works.
It would be good to have a clean solution.