nanoframework/Home #1219: It cannot to find the extension

ArBom, 2023-01-30:
### Tool

Visual Studio extension

### Description

There is imposible to create new nanoFramework project (Blank Application) at VS2017 after install this framework. Is shows "Issue of packet installation" window. I tried to install this framework from marketplace a few times.

[image]

### Aditional context

VS Enterprise 2017 15.9.51
nanoFramework 2017.8.0.1

torbacz, 2023-01-30:
VS2017 is not supported according to official guide (only VS2019 and VS2022), please refer to https://docs.nanoframework.net/content/getting-started-guides/getting-started-managed.html

josesimoes, 2023-01-30:
The above is correct. I've discontinued the VS 2017 extension a long time ago. Please use VS2019 or, preferably, VS 2022.

closed 2023-01-30 by josesimoes