nanoframework/Home #1223: ** Socket exception occurred: Exception was thrown: System.Net.Sockets.SocketException error code 10065!**

zhangqiang199226, 2023-02-09:
### Library/API/IoT binding

WifiAP Example And Wifi Example

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.2.0.48

### Target name(s)

ESP32_PSRAM_REV3

### Firmware version

1.8.0.973

### Device capabilities

Use the ESP32 development board to use wifiAP to do the Socket server, and wifi to do the Socket client connection error: ** Socket exception occurred: Exception was thrown: System.Net.Sockets.SocketException error code 10065!**

### Description

Use the ESP32 development board to use wifiAP to do the Socket server, and wifi to do the Socket client connection error: ** Socket exception occurred: Exception was thrown: System.Net.Sockets.SocketException error code 10065!**

### How to reproduce

1.   Client :    mySocket.Connect(iPEndPoint);

                    Debug.WriteLine("Connecting...");

                    byte[] buffer = Encoding.UTF8.GetBytes("Hello World!");

                    mySocket.Send(buffer);

                    Debug.WriteLine($"Wrote {buffer.Length} bytes");

                    byte[] bytes = new byte[200];
                    int num = mySocket.Receive(bytes);
                    Debug.WriteLine(Encoding.UTF8.GetString(bytes,0,num));
2.Server  :  serviceSocket = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp);
            Debug.WriteLine(WirelessAP.GetIP());
            serviceSocket.Bind(new IPEndPoint(IPAddress.Parse(WirelessAP.GetIP()),2000));
            
            serviceSocket.Listen(10);

            while (true)
            {
                Debug.WriteLine("！");
                Socket client = serviceSocket.Accept();
                IPEndPoint iPEndPoint = client.RemoteEndPoint as IPEndPoint;

                try
                {
                    if (iPEndPoint != null)
                    {
                        int num = 0;
                        for (int i = 0; i <clients.Length - 1; i++)
                        {
                            if (string.IsNullOrEmpty(clients[i]))
                            {
                                num = i;
                                break;
                            }
                        }
                        clients[num] = iPEndPoint.Address.ToString() + " " + iPEndPoint.Port.ToString();
                        sockets.Add(client);
                        s_hashtable.Add(clients[num], sockets[num]);

                        ThreadWithState withState = new ThreadWithState(client, s_hashtable,clients,sockets);

                        Thread thread = new Thread(new ThreadStart(withState.ThreadProc));
                        thread.Start();
                    }
                }
                catch (Exception)
                {

                }
            }

### Expected behaviour

Socket server Connetion to Socket client.

### Screenshots

[image]

### Aditional information

1.   Client :    mySocket.Connect(iPEndPoint);

                    Debug.WriteLine("Connecting...");

                    byte[] buffer = Encoding.UTF8.GetBytes("Hello World!");

                    mySocket.Send(buffer);

                    Debug.WriteLine($"Wrote {buffer.Length} bytes");

                    byte[] bytes = new byte[200];
                    int num = mySocket.Receive(bytes);
                    Debug.WriteLine(Encoding.UTF8.GetString(bytes,0,num));
2.Server  :  serviceSocket = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp);
            Debug.WriteLine(WirelessAP.GetIP());
            serviceSocket.Bind(new IPEndPoint(IPAddress.Parse(WirelessAP.GetIP()),2000));
            
            serviceSocket.Listen(10);

            while (true)
            {
                Debug.WriteLine("！");
                Socket client = serviceSocket.Accept();
                IPEndPoint iPEndPoint = client.RemoteEndPoint as IPEndPoint;

                try
                {
                    if (iPEndPoint != null)
                    {
                        int num = 0;
                        for (int i = 0; i <clients.Length - 1; i++)
                        {
                            if (string.IsNullOrEmpty(clients[i]))
                            {
                                num = i;
                                break;
                            }
                        }
                        clients[num] = iPEndPoint.Address.ToString() + " " + iPEndPoint.Port.ToString();
                        sockets.Add(client);
                        s_hashtable.Add(clients[num], sockets[num]);

                        ThreadWithState withState = new ThreadWithState(client, s_hashtable,clients,sockets);

                        Thread thread = new Thread(new ThreadStart(withState.ThreadProc));
                        thread.Start();
                    }
                }
                catch (Exception)
                {

                }
            }

cbagpipe, 2023-02-22:
How do you manage new AP clients? Does they properly connected and has same subnet as AP?

zhangqiang199226, 2023-02-22:
> How do you manage new AP clients? Does they properly connected and has same subnet as AP?

One ESP32 does the AP and then opens the Socket server, and the other ESP32 does the same network segment Socket client as the wifi client connects to the AP server to report this error. There is no problem connecting with a PC.

closed 2023-04-13 by zhangqiang199226