nanoframework/Home #1225: Need images for ESP32_C3 with internal USB connection

josesimoes, 2023-02-16:
### Description

Our current build for ESP32_C3 is configured to use UART0 for communication with the board. Some hardware like the Xiao ESP32C3 is using the internal USB hardware.

### How to solve the problem

Need a new build to use the internal USB hardware to allow this to be used in such devices.

matsujirushi, 2023-02-28:
Can I try the implementation?
It is the first time to add a function to nanoFramework.

josesimoes, 2023-02-28:
Hey @matsujirushi glad that you've offered for this!! 👏🏻 
Just to properly setup expectation: making this happen it's strictly native code (C) and tweaking the build system (CMake).

Please confirm.

matsujirushi, 2023-02-28:
Hi @josesimoes ,
Thanks for message.
I can write C/C++ and  a little CMake.
I would like to try it.

matsujirushi, 2023-03-09:
FYI

I finally got it working.
still unstable. It looks like it will take time.

[image]

https://github.com/matsujirushi/nf-interpreter/commits/develop-esp32c3-cdcacm

matsujirushi, 2023-03-13:
Currently, I have two issues.

1. Visual Studio can no longer communicate with XIAO ESP32C3 after stopping debugging.
  It seems that the USB interface is not sufficiently reset by the Monitor_Reboot command.
2. Device resets on first COM connection.
  The device is resetting because Visual Studio is signaling RTS=1, DTR=0. It seems necessary to change this. ->  [System.IO.Ports](https://github.com/dotnet/runtime/blob/541347cbb4270f3861a8bb5eada9f1547d309b3a/src/libraries/System.IO.Ports/src/System/IO/Ports/SerialStream.Windows.cs#L1148)

josesimoes, 2023-03-13:
> 1. Visual Studio can no longer communicate with XIAO ESP32C3 after stopping debugging.
>    It seems that the USB interface is not sufficiently reset by the Monitor_Reboot command.

If that's the case, then it's better to perform a hard reset instead of "just" a CLR reboot.

> 2. Device resets on first COM connection.
>    The device is resetting because Visual Studio is signaling RTS=1, DTR=0. It seems necessary to change this. 

We can (should?) change the configuration of the serial port. It seems a bit harsh that this requires a change on the .NET library...

matsujirushi, 2023-03-13:
> If that's the case, then it's better to perform a hard reset instead of "just" a CLR reboot.

The nanoCLR restarted the whole with esp_restart(), but it seems that the USB part was not reset well. 😣
Create an issue to ESP-IDF team.

> It seems a bit harsh that this requires a change on the .NET library...

I can only think of a way to change the .NET SerialPort or call it directly with the Win32 API.

The first issue is serious, so I'll focus on it.

josesimoes, 2023-03-13:
@matsujirushi sound good. Please either refer the IDF issue from here so this can be track and traced, if needed.

matsujirushi, 2023-03-18:
much better now.
https://github.com/matsujirushi/nf-interpreter/commits/develop-esp32c3-cdcacm
https://github.com/matsujirushi/nf-interpreter/releases/tag/preview

@josesimoes 
It would be nice to be able to switch between UART and USB_SERIAL_JTAG at compile time.
How would it be better to allow it to be specified?

josesimoes, 2023-03-20:
There is already an HAL_WP_SERIAL build option 

https://github.com/nanoframework/nf-interpreter/blob/main/targets/AzureRTOS/SiliconLabs/SL_STK3701A/CMakePresets.json#L46

This is implemented in Azure RTOS and Silabs. It would have to be implemented in ESP32.

I have to say that I'm not very fond of the serial_Jtag name. It can be misleading. That may come from IDF, but it's not a good name here.

matsujirushi, 2023-03-20:
Thank you for advise.
I would like to be able to specify it with `HAL_WP_USE_USB_CDC`.

https://github.com/nanoframework/nf-interpreter/blob/main/targets/AzureRTOS/SiliconLabs/SL_STK3701A/CMakePresets.json#L62-L63

matsujirushi, 2023-03-21:
Updated.
https://github.com/matsujirushi/nf-interpreter/tree/develop-esp32c3-cdcacm
https://github.com/matsujirushi/nf-interpreter/releases/tag/preview2

@josesimoes 
I would like to create a pull request.
Do you have any concerns?

josesimoes, 2023-03-21:
@matsujirushi please go for it! 🦾

closed 2023-04-27 by josesimoes