nanoframework/Home #1228: Non-empty classes serialized with `BinaryFormatter` on NanoFramwrork incompatible with ones on .NET

cbagpipe, 2023-02-21:
### Library/API/IoT binding

System.Runtime.Serialization, nanoFramework.Serialization.Helper

### Visual Studio version

VS2022 (17.4.4)

### .NET nanoFramework extension version

2022.3.0.19

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.1043

### Device capabilities

```
System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 6407ecb
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Feb 15 2023
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.1043
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Teleavtomatika.IoT.Nanoframework.TestApp, 1.0.0.0
  System.Diagnostics.Stopwatch, 1.2.0.0
  mscorlib, 1.14.3.0
  Windows.Storage, 1.5.33.0
  System.Net.Sockets.UdpClient, 1.1.43.0
  Windows.Storage.Streams, 1.14.24.46541
  System.Device.Gpio, 1.1.28.0
  System.Runtime.Serialization, 1.0.20.0
  nanoFramework.System.Text, 1.2.37.0
  System.Math, 1.5.29.0
  System.Net.Sockets.TcpClient, 1.1.52.52655
  nanoFramework.System.Collections, 1.5.18.0
  nanoFramework.Hardware.Esp32, 1.5.1.0
  Iot.Device.Uln2003, 1.2.0.0
  System.IO.FileSystem, 1.1.23.0
  System.Drawing, 1.2.0.0
  System.Net, 1.10.52.0
  nanoFramework.Runtime.Native, 1.6.6.0
  Iot.Device.Ws28xx, 1.2.0.0
  System.IO.Streams, 1.1.38.0
  System.Threading, 1.1.19.33722
  System.Device.Wifi, 1.5.51.0
  nanoFramework.Json, 2.2.84.0
  nanoFramework.Runtime.Events, 1.11.6.0
  System.Device.Spi, 1.3.37.0
  Teleavtomatika.IoT.Nanoframework.Protocol, 1.0.0.0
  Teleavtomatika.IoT.Nanoframework.StepperMotorShutter, 1.0.0.0
  Teleavtomatika.IoT.Nanoframework.Core, 1.0.0.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty
```

### Description

Empty class, shared between nanoFramework Device and .NET app, and serialized with `BinaryFormatter` has same type CRC
But class with properties not. So `System.Runtime.Serialization.Formatters.Binary.BinaryFormatter` becomes totally incompatible with `nanoFramework.Serialization.Helper.BinaryFormatter`

### How to reproduce

1. Create new solution with two projects: one nanoFramework App and one NET console App
2. Create file with any class
3. Add property to that class
4. Link that file to both projects
5. Try to serialize file with BinaryFormatter in both projects
6. See that first 4 bytes of resulting arrays (where calculated type hashes stored) differs one from another.

### Expected behaviour

Same hashes on both platforms

### Sample project or code

https://github.com/cbagpipe/BinaryFormatterIssue

alberk8, 2023-03-12:
I have also tested on ESP32 and using the unit test data on ESP32 it does not match.   Here is the Sample [Code](https://github.com/alberk8/nanoFramework-SerializationESP32)

josesimoes, 2023-03-13:
The test code used here is missing a new line to instantiate a new `BinaryFormatter()` before serializing the second class.
Serialization occurs by sweeping the class data and producing an unique binary output. When you start serializing the second class reusing the same serializer helper it picks up where it ended, thus producing "wrong" output. 

This can be made clear in the documentation and also in the code by adding a flag that throws an `InvadliOperation` exception  if the `Serialize()` method is called a second time.

Any chance you can take of these improvements?

cbagpipe, 2023-03-13:
Test project code updated. Result — same: nanoFramework serialized bytes different from NET serialized bytes.

Ellerbach, 2023-04-17:
Confirmed that running the unit tests on an ESP32 fails at all the serialization:
[image]

So having a look at all this

Ellerbach, 2023-04-17:
I can confirm as well that the behavior is different depending on the target used.

I took this code:

```csharp
            var empty = new EmptyClass();
            var eBytes = BinaryFormatter.Serialize(empty);
            var eBytesString = BitConverter.ToString(eBytes);
            Debug.WriteLine(eBytesString); // 82-A9-1C-33-C0

            var notEmpty = new NotEmptyClass();
            var neBytes = BinaryFormatter.Serialize(notEmpty);
            var neBytesString = BitConverter.ToString(neBytes);
            Debug.WriteLine(neBytesString); // BC-EB-EE-C8-80-00-00-00-00
```

from the initial issue.

Run on an STM32:

```text
82-A9-1C-33-C0
9F-91-7E-2A-80-00-00-00-00
```

ESP32-C3 or ESP32:

```text
82-A9-1C-33-C0
BC-EB-EE-C8-80-00-00-00-00
```

So there is defintely something broken somewhere.

networkfusion, 2023-04-17:
Thoughts here would include the following possibilities:

* Default UTF conversion
* LSB vs MSB

alberk8, 2023-04-18:
Could be the hashing done on the native that is the cause?  I tried to trace that but got loss in to bowels of CLR. :(

Ellerbach, 2023-04-18:
Yes, it can be all this at the same time. And the point is that it's different per platform so far. But the good news is that it's symetrical, so it deserialize properly what it does serialize regardless of the platform!

I ran it also in the Virtual Device (the same used for the tests) and yet another result:

```text
82-A9-1C-33-C0
8E-9C-2C-7C-80-00-00-00-00
```

So definitely somethis on the native side that is not properly set. It's going to take a lot of time for the investigation as it's quite complex.

Ellerbach, 2023-04-18:
The empty class always serialize properly but the other one with 1 property does not. Looking at the differences:

```text
Win32: 1000 1110 1001 1100 0010 1100 0111 1100
ESP32: 1011 1100 1110 1011 1110 1110 1100 1000
STM32: 1001 1111 1001 0001 0111 1110 0010 1010
```
Looking at the pattern here, it seems that there is more than a LSB/MSB issue.

Ellerbach, 2023-05-03:
> This seems to be pointing to something platform specific with ESP32.

Still with the PybStick, I don't have the same result as Windows neither ESP32. So definitely, different series can produce a different result from what I can see with my devices.

josesimoes, 2023-05-09:
Fixed through nanoframework/nf-interpreter#2656.

closed 2023-05-09 by josesimoes