nanoframework/Home #1229: Helper Method to Re-Format Internal Flash Drives on Boards that Support It

gligorov, 2023-02-22:
### Description

At present there is no way to reformat the internal flash on devices.  If for any reason, format is needed, ( security, troubleshooting etc.), there is no way to do that.

### How to solve the problem

Add a simple helper method like  Format(string driveLetter) or something similar to be used from a library that is best suited for this.

### Describe alternatives you've considered

none at present

josesimoes, 2023-04-19:
Nope. This requires a native call in order to restart/format the file system or SPIFFs. When it comes to this, it's because the file system got unusable.

torbacz, 2023-12-29:
Maybe command with nanoff will be sufficient?

networkfusion, 2024-02-05:
@josesimoes should this be closed now?

josesimoes, 2024-02-06:
> @josesimoes should this be closed now?

The linked PR has added support for this only for SPIFFs storage. I think it should be extended to the other supported storage: SD Cards and USB MSD. That's why I left it open.

closed 2024-07-23 by josesimoes