nanoframework/Home #1235: [FeatherS2] PWM can't be properly created

Ellerbach, 2023-03-03:
### Discussed in https://github.com/nanoframework/Home/discussions/1230

<div type='discussions-op-text'>

<sup>Originally posted by **sangyuxiaowu** February 24, 2023</sup>
Hi,
I'm trying to use my board to control the servos, according to this article https://docs.nanoframework.net/devicesdetails/ServoMotor/README.html

```
Configuration.SetPinFunction(2, DeviceFunction.PWM16);
PwmChannel pwmChannel = PwmChannel.CreateFromPin(2, 50);
ServoMotor servoMotor = new ServoMotor(pwmChannel);
servoMotor.Start();

servoMotor.WritePulseWidth(0);
Thread.Sleep(2000);
servoMotor.WritePulseWidth(90);
Thread.Sleep(2000);
servoMotor.WritePulseWidth(180);
Thread.Sleep(Timeout.Infinite);
```

I'm sure there is no problem with the wiring, but this error is reported when creating the PwmChannel. It seems that it cannot be set to 50. After testing, it is found that it needs to be set to at least 2432

```text
引发的异常:“System.ArgumentException”(位于 System.Device.Pwm.dll 中)
    ++++ Exception System.ArgumentException - 0xfd000000 (1) ++++
    ++++ Message: 
    ++++ System.Device.Pwm.PwmChannel::NativeSetDesiredFrequency [IP: 0000] ++++
    ++++ System.Device.Pwm.PwmChannel::set_Frequency [IP: 0005] ++++
    ++++ System.Device.Pwm.PwmChannel::CreateFromPin [IP: 0026] ++++
    ++++ MotionControl.Program::Main [IP: 0039] ++++
```

board brushed FEATHER_S2 firmware. servos is SG92R
information:
https://www.waveshare.com/wiki/ESP32-S2-Pico

It's a strange question, because I'm just a hobbyist, I may not know some basic knowledge.

Any help at all appreciated!

thanks

</div>

Ellerbach, 2023-03-03:
See discussion https://github.com/nanoframework/Home/discussions/1230

josesimoes, 2024-06-12:
Can you please confirm that this has been fixed by now?

josesimoes, 2025-02-05:
@Ellerbach can you please confirm if this has been fixed (because I think it was)?

networkfusion, 2025-04-06:
For reference, the FEATHER_S2 is now ESP32_S2_USB.