nanoframework/Home #1236: MANUFACTURER SPECIFIC DATA can't be added when start a advertising

jeffer80, 2023-03-03:
### Library/API/IoT binding

nanoFramework.Device.Bluetooth

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.3.0.9

### Target name(s)

ESP32_BLE_REV3

### Description

When start advertising for a service, there is no way to add the manufacturer special data to the advertising data.
serviceProvider.StartAdvertising(new GattServiceProviderAdvertisingParameters()
            {
                DeviceName = "BLE Device",
                IsConnectable = true,
                IsDiscoverable = true,
            });

josesimoes, 2023-03-03:
@AdrianSoundy comments on this please?

AdrianSoundy, 2023-03-04:
Will look into this issue

AdrianSoundy, 2023-03-11:
Will change this to include the missing elements in advertisments.
- manufacturers data
- appearance
- service data

axa88, 2023-03-13:
In the latest release of the Bluetooth library, the `DeviceName` property was removed from the `GattServiceProviderAdvertisingParameters`, and now the device name can be set via the `BluetoothLEServer` object.

I am unfamiliar with what framework nf is trying to follow but when investigating I find that 
[GattServiceProviderAdvertisingParameters ](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattserviceprovideradvertisingparameters?view=winrt-22621)(in WinRT at least), only exposes `IsConnectable` `IsDiscoverable` `ServiceData`

Therefore I did not report this as a problem believing nf is trying to follow the windows sdk.
So my question is, under what justification are other things to be added as was proposed above: 
```
    manufacturers data
    appearance
    service data
```
PS. My question is as much an inquiry about the ble spec as it is about aligning the sdk. I simply dont have a frame of reference.

AdrianSoundy, 2023-03-13:
The latest release aligned thngs to be closer to the Windows.device.bluetooth by removing the devcie name from advertisment and removing the Gattserviceprovider.addservice. 
I had to add the BlutoothLEServer class to do all the things that are done automatically or by GUI (manually) by windows implementation.

The advertistment data should follow closely to the the Windows.Device.framework. Adding anything that is missing from Windows that we require.
I am currently working on the generic access parts for server / client and should soon be starting on this issue.

jeffer80, 2023-03-14:
> In the latest release of the Bluetooth library, the `DeviceName` property was removed from the `GattServiceProviderAdvertisingParameters`, and now the device name can be set via the `BluetoothLEServer` object.
> 
> I am unfamiliar with what framework nf is trying to follow but when investigating I find that [GattServiceProviderAdvertisingParameters ](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattserviceprovideradvertisingparameters?view=winrt-22621)(in WinRT at least), only exposes `IsConnectable` `IsDiscoverable` `ServiceData` (The latter of which is already contained in the object) Therefore I did not report this as a problem believing nf is trying to follow the windows sdk.
> 
> So my question is, under what justification are other things to be added as was proposed above:
> 
> ```
>     manufacturers data
>     appearance
>     service data
> ```
> 
> PS. My question is as much an inquiry about the ble spec as it is about aligning the sdk. I simply dont have a frame of reference.

below is the quote from "Supplement to the Bluetooth Core Specification | CSS v10, Part A, Section 1.4"

1.4 MANUFACTURER SPECIFIC DATA
1.4.1 Description
The Manufacturer Specific data type is used for manufacturer specific data.
The first two data octets shall contain a company identifier from Assigned
Numbers. The interpretation of any other octets within the data shall be defined
by the manufacturer specified by the company identifier.

axa88, 2023-03-15:
> below is the quote from "Supplement to the Bluetooth Core Specification | CSS v10, Part A, Section 1.4"

I see, and ill assume the table on page 10 specifies which of the types should be part of advertising data.
It would be great to see the `TX Power Level` part of the advertisement as well, then we can start using it for proximity when combined with rssi.
@AdrianSoundy  what do you think about that?

AdrianSoundy, 2023-04-16:
This issue will be fixed when the changes to advertisement are completed(in progress)
The original GattServiceProviderAdvertisingParameters didn't allow for the extra advertisement properties to be changed.

The next changes will include the BluetoothLEAdvertisement class and BluethLEAdvertisementPubliser which allows access to all prooperties. The adverticement in the Service Provider has been changed to use the BluetoothLEAdvertisement & BluethLEAdvertisementPubliser internally and the GattServiceProviderAdvertisingParameters chnaged to give access to the  BluetoothLEAdvertisement object so any property can be added. The BluethLEAdvertisementPubliser  can be used to create advertisements for ibeacons.

jeffer80, 2023-04-20:
> This issue will be fixed when the changes to advertisement are completed(in progress) The original GattServiceProviderAdvertisingParameters didn't allow for the extra advertisement properties to be [...]

perfect, waiting for your good news

closed 2023-08-02 by josesimoes