nanoframework/Home #1240: Bug report for documentation

DaveSchmid, 2023-03-14:
### Description

One should have the possibility to report a bug in the documentation.

### How to solve the problem

A choice in
https://github.com/nanoframework/Home/issues/new/choose

josesimoes, 2023-03-14:
@DaveSchmid all the issue reporting and work is tracked here on a single repository, by design. To keep things centralized as the project (and issues) often cross several domains and repositories, meaning that we would have duplicated or mirror issues. 

On the documentation site specifically, on each page there is a link at the top right "IMPROVE THIS DOC" that takes you immediately to the source file where you can edit right on the web and start a PR with zero effort. Does that help?

josesimoes, 2023-03-14:
If that's a typo or a formatting error you just need to edit _that_ document, submit the PR with the changes, and that will fix it.
If it is something else, please explain.

DaveSchmid, 2023-03-14:
There is an error in a graphic. It's easy once you have the template and the specific program that created the graphic. That's why I just want to point out the error without fixing it myself. There is no corresponding option for this.

josesimoes, 2023-03-14:
The source of the images is not on github. Maybe we should improve that. At this point you'll have to jot a print screen to point it out..

DaveSchmid, 2023-03-14:
I've already done that, but when I want to create an issue, there's no option for that kind of problem.

DaveSchmid, 2023-03-14:
The "Issue: Bug report for libraries, IoT bindings or API" isn't really the right place for it.

commit f1a32f73 (2023-04-13): Add template for docs issues

- Closes #1240.
commit 2640d3a8 (2023-04-13): Add template for docs issues

- Closes #1240.
closed 2023-04-13 by josesimoes