nanoframework/Home #1241: Error in documentation graphics

DaveSchmid, 2023-03-14:
### Library/API/IoT binding

nanoframework.github.io/images/architecture.png

### Description

The abbreviation for "Platform Abstraction Layer" is PAL not HAL.

[image]

Ellerbach, 2023-03-14:
Great catch :-D PR Welcome ;-) Otherwise, this will wait for one of the next PR on maintenance for the docs.

DaveSchmid, 2023-03-14:
A small bug if you have the program and template for the picture. Otherwise a bit difficult.

closed 2023-03-14 by Ellerbach