nanoframework/Home #1242: 404 error in documentation generation

DaveSchmid, 2023-03-14:
### Library/API/IoT binding

Menu of docs.nanoframework.net

### Description

Clicking in the menu "Documentation -> HAL and PAL" produces a 404 error.

[image]

[image]

Ellerbach, 2023-03-14:
Thanks @DaveSchmid, problem identified and will be fixed. It's related to a regression in one of the tool used for the automation.
Regression fixed in the tool, PR pushed on that side and approved, released on its way. As soon as choco is up to date, I'll rerun the last pipeline and you should be able to enjoy that page :-)
If not, then, I'll take another look at this problem!

DaveSchmid, 2023-03-14:
Thanks @Ellerbach

closed 2023-03-14 by Ellerbach