nanoframework/Home #1244: ESP32 reboot during Initialization

alberk8, 2023-03-17:
### Target name(s)

ESP32_PSRAM_REV3

### Firmware version

1.8.1.65

### Was working before? On which version?

N/A

### Description

Occasionally during reboot (pressing the reboot/reset button) or during Cold Boot.   The ESP32 will reboot twice.  I capture that is from COM port output for both ESP32_PSRAM_REV3 as well as M5Stack Core2.

### How to reproduce

It cannot be consistently reproduces.  It only happen once in a while.

### Expected behaviour

Reboot only happen once.

### Aditional information

```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: BREAKN instr
Core  1 register dump:
PC      : 0x401e6f4f  PS      : 0x00060736  A0      : 0x800d338e  A1      : 0x3ffde8c0
A2      : 0x0000001c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000003
A6      : 0x00000001  A7      : 0x3ffde8b0  A8      : 0x800d7d9d  A9      : 0x3ffde5f0
A10     : 0x3f800888  A11     : 0x000000cb  A12     : 0x3fae8bb4  A13     : 0x000003a4
A14     : 0x00000001  A15     : 0x3ffde5f0  SAR     : 0x00000018  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x401e6f5c:0x3ffde830 0x400d338b:0x3ffde850 0x400ddf21:0x3ffde8d0 0x400ddf9d:0x3ffde920 0x400de696:0x3ffde960 0x400da5ab:0x3ffde9b0 0x400e1ef5:0x3ffdea20 0x400e1cd9:0x3ffdea60 0x400dcc55:0x3ffdeaa0 0x400dc3dd:0x3ffdeae0 0x400f4f29:0x3ffdeb40 0x40129166:0x3ffdeb80 0x401273c5:0x3ffdeba0
```

**BackTrace Decoded**
```
0x401e6f5c: HAL_AssertEx at /workspaces/nf-interpreter/targets/ESP32/_common/targetHAL.c:11
0x400d338b: CLR_RT_HeapBlock::ObjectsEqual(CLR_RT_HeapBlock const&, CLR_RT_HeapBlock const&, bool) at /workspaces/nf-interpreter/src/CLR/Core/CLR_RT_HeapBlock.cpp:1387
0x400ddf21: CLR_RT_ExecutionEngine::FindLockObject(CLR_RT_DblLinkedList&, CLR_RT_HeapBlock&) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:2319
0x400ddf9d: CLR_RT_ExecutionEngine::FindLockObject(CLR_RT_HeapBlock&) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:2355
0x400de696: CLR_RT_ExecutionEngine::LockObject(CLR_RT_HeapBlock&, CLR_RT_SubThread*, long long const&, bool) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:2689
0x400da5ab: CLR_RT_StackFrame::HandleSynchronized(bool, bool) at /workspaces/nf-interpreter/src/CLR/Core/CLR_RT_StackFrame.cpp:803
0x400e1ef5: CLR_RT_Thread::Execute_Inner() at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:785
0x400e1cd9: CLR_RT_Thread::Execute() at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:642
0x400dcc55: CLR_RT_ExecutionEngine::ScheduleThreads(int) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:1207
0x400dc3dd: CLR_RT_ExecutionEngine::Execute(wchar_t*, int) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:634
0x400f4f29: ClrStartup at /workspaces/nf-interpreter/src/CLR/Startup/CLRStartup.cpp:405
0x40129166: CLRStartupThread at /workspaces/nf-interpreter/targets/ESP32/_nanoCLR/CLR_Startup_Thread.c:17
0x401273c5: main_task at /workspaces/nf-interpreter/targets/ESP32/_IDF/esp32/app_main.c:37

```

josesimoes, 2023-03-17:
No idea what could be causing this, but your trace there sure it's helpful when this will be looked at.

Ellerbach, 2023-04-18:
I've seen this time to time with the Core2. Having a powerfull external power seems to fix the issue for me. It happend mainly when the screen was switching on and trying to debug.

josesimoes, 2023-05-29:
@alberk8 is this still happening with the latest version?

josesimoes, 2024-06-12:
@alberk8 is this still happening? Haven't seen any similar reports and ESP32 have been under heavy testing, so I supposed this has been fixed...

alberk8, 2024-06-12:
Nothing so far with the IDF 5.1.x.  So closing it.

closed 2024-06-12 by alberk8