nanoframework/Home #1247: Get Connected AP information

alberk8, 2023-03-18:
### Description

Able to get the Connected AP information like BSSID, SSID,  RSSI, Channel, Cipher, Phy, etc.  ESP32 native [code](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?highlight=esp_wifi_sta_get_ap_info#_CPPv416wifi_ap_record_t)

### How to solve the problem

Ability to get information on the connected AP at regular interval.

Ellerbach, 2023-04-18:
I guess it's something that can be done. It would require the same elements or at least a large part of the subsets on the other platforms than ESP32.
Setting up a low priority for this one and up to grab!

networkfusion, 2025-03-07:
Potentially linked: https://github.com/nanoframework/Home/issues/1335