nanoframework/Home #1258: Cannot find source file - System.Device.I2s_SRC_FILE-NOTFOUND

matsujirushi, 2023-03-29:
### Target name(s)

FEATHER_S2

### Firmware version

1207719194a556b8d58e89adc9d538b70ad339d3

### Description

An error occurs in CMake when building nanoCLR for FEATHER_S2 in a devcontainer environment.

```
[cmake] -- Configuring done
[cmake] CMake Error at CMake/Modules/FindNF_NativeAssemblies.cmake:423 (add_library):
[cmake]   Cannot find source file:
[cmake] 
[cmake]     System.Device.I2s_SRC_FILE-NOTFOUND
[cmake] Call Stack (most recent call first):
[cmake]   CMake/binutils.ESP32.cmake:260 (nf_add_lib_native_assemblies)
[cmake]   CMake/binutils.ESP32.cmake:414 (nf_add_platform_dependencies)
[cmake]   targets/ESP32/ESP32_S2/CMakeLists.txt:10 (nf_setup_target_build)
[cmake] 
[cmake] 
[cmake] CMake Error at CMake/Modules/FindNF_NativeAssemblies.cmake:423 (add_library):
[cmake]   No SOURCES given to target: NF_NativeAssemblies
[cmake] Call Stack (most recent call first):
[cmake]   CMake/binutils.ESP32.cmake:260 (nf_add_lib_native_assemblies)
[cmake]   CMake/binutils.ESP32.cmake:414 (nf_add_platform_dependencies)
[cmake]   targets/ESP32/ESP32_S2/CMakeLists.txt:10 (nf_setup_target_build)
```

### Aditional information

Case doesn't match.

[image]

matsujirushi, 2023-03-29:
Please assign me. I will create PR.

josesimoes, 2023-03-29:
@matsujirushi thanks for the detailed report.

We always try to keep things as simple as possible and don't waste people valuable time.
You could have go straight to the PR. No need for an issue.
In this case as the issue was investigated and the fix was agreed.